KalicoCrew/kalico #49: moonraker connection issues

dans98, 2023-08-27:
I'm try to switch to danger klipper from klipper, so i can do some testing and do a pr for the pid calibration improvements. I used kiauh to switch over to my fork of danger clipper master, but im getting an error thats looks to be encoding or json decoding related.

klipper: v0.0.0-5104-g2c0d70bb-inferred
mainsail: v2.7.1
moonraker: v0.8.0-138-gfe120952

from the klipper log
```
webhooks: Error decoding Server Request {"id": 1978684400, "method": "info", "params": {"client_info": {"program": "Moonraker", "version": "v0.8.0-138-gfe12095"}}}
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/webhooks.py", line 268, in process_received
    web_request = WebRequest(self, req)
  File "/home/pi/klipper/klippy/webhooks.py", line 63, in __init__
    raise ValueError("Invalid request type")
ValueError: Invalid request type

```

from the moonraker log
```
2023-08-27 15:42:55,577 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: printer.info, Code: -32601, Message: Method not found
2023-08-27 15:42:57,577 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: printer.info, Code: -32601, Message: Method not found
2023-08-27 15:42:59,579 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: printer.info, Code: -32601, Message: Method not found
2023-08-27 15:43:01,578 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: printer.info, Code: -32601, Message: Method not found
2023-08-27 15:43:03,582 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: printer.info, Code: -32601, Message: Method not found
2023-08-27 15:43:04,831 [klippy_connection.py:wait()] - Request 'info' pending: 60.00 seconds
```

logs:
[klippy.log](https://github.com/DangerKlippers/danger-klipper/files/12448822/klippy.log)
[moonraker.log](https://github.com/DangerKlippers/danger-klipper/files/12448823/moonraker.log)

i went into the WebRequest __init__ and added a little logging.
```python
        if not isinstance(self.method, str) or not isinstance(
            self.params, dict
        ):
            logging.info("dls method is : %s" % (type(self.method)))
            logging.info("dls params is : %s" % (type(self.params)))
            raise ValueError("Invalid request type")
```
the logged info.
```
dls method is : <type 'unicode'>
dls params is : <type 'dict'>

```

I'm not sure why this is happening, as i thought unicode was considered to be a string. if i revert to klipper everything works fine.

bwnance, 2023-08-27:
@dans98 this is really weird, but i had the same issue recently, i'll dig into it and see what's up

rogerlz, 2023-08-27:
I would say that python2 is probably the reason
`Python: '2.7.16 (default, Oct 10 2019, 22:02:15) \n[GCC 8.3.0]'`

bwnance, 2023-08-27:
oh yeah - dangerklipper drops support for python2

dans98, 2023-08-27:
hmm, let me see if i can get to 3 without having to completely rebuild the pi from scratch.

bwnance, 2023-08-27:
@dans98 what you could do is use kiauh to uninstall klipper, then use kiauh to install it with python3

dans98, 2023-08-28:
@bwnance I tried thats but it didn't work. 

I tried completely clean slate install last night but ran into what i assume is the debian usb bug.

I'm going to try a clean state setup again today and then run the following commands.
```
sudo ufw disable
sudo apt remove brltty
sudo systemctl disable ModemManager
curl -sSL https://raw.githubusercontent.com/mainsail-crew/MainsailOS/develop/patches/udev-fix.sh | bash
sudo reboot
```

Seems it's the recommended way to deal with the debian issue on the klipper discord.

bwnance, 2023-08-28:
@dans98 ok, thank you! we'll definitely take a look at why this is going on - we'd like the barrier of entry for danger klipper to be minimal 😅

dans98, 2023-08-28:
@bwnance i think switching to python 3 will fix the above initial issue.  I think I'm running into the debian usb bug now, because after i flash the mcu it doesn't show up anymore.  I think i got lucky in the past, as i started with mainsailOs like 20 months ago and everything just worked!

bwnance, 2023-08-28:
oh, ok! let me know if I can help with anything!

rogerlz, 2023-09-04:
@dans98 did you sort it out? anything I can help you with?

dans98, 2023-09-04:
@rogerlz 

Sorry I had family in town for a few days so I haven't had a chance to respond. 

It took a few tires, but it seems none of the 'fixes' work with my duet 2.  What worked was going back to mainsailOS version 0.7.1 (the last buster build), and then just upgrading everything. That got me to python 3.7.3, and the latest releases of klipper, mainsail, and moonraker.

when my family leaves tomorrow i'm hoping to get converted over to danger klipper. I'll post an update!