KalicoCrew/kalico #63: erratic PID tuning

kjkent, 2023-09-28:
Hi there. I've been using @dans98 's fantastic PID tuning improvements on my Artillery Genius printer since moving it over to Klipper. The improvements have made Klipper useable for me in terms of fixing bed temp overshoot.

I've recently replaced my hotend heat block, and my hotend temps have been overshooting pretty severely since. I did a PID tune using stock Klipper, and gave up waiting for the integral windup overshoot to settle to the target. So, retrying the updated PID tuning code, I get results that I don't recall as typical, based on my bed tuning escapades last year.

Here's an example:

```
01:50 PID parameters: pid_Kp=63.194 pid_Ki=7.435 pid_Kd=134.288
The SAVE_CONFIG command will update the printer config file
with these parameters and restart the printer.
01:50 sample:15 pwm:1.0000 asymmetry:-5.1549 tolerance:0.0000
01:50 sample:14 pwm:1.0000 asymmetry:9.4129 tolerance:14.9269
01:50 sample:13 pwm:1.0000 asymmetry:-0.1686 tolerance:14.9269
01:50 sample:12 pwm:1.0000 asymmetry:-6.5073 tolerance:14.9269
01:49 sample:11 pwm:-13.9269 asymmetry:-3.4528 tolerance:14.9269
01:49 sample:10 pwm:-9.3790 asymmetry:2.5606 tolerance:10.3790
01:49 sample:9 pwm:-1.9767 asymmetry:4.6231 tolerance:5.4575
01:49 sample:8 pwm:1.0000 asymmetry:4.8241 tolerance:5.4575
01:49 sample:7 pwm:1.0000 asymmetry:5.4395 tolerance:5.4575
01:49 sample:6 pwm:-4.4575 asymmetry:-4.5732 tolerance:5.2660
01:49 sample:5 pwm:-3.4905 asymmetry:0.9791 tolerance:4.4905
01:49 sample:4 pwm:-0.9058 asymmetry:-5.9395 tolerance:1.9058
01:49 sample:3 pwm:0.8085 asymmetry:-5.1476 tolerance:n/a
01:49 sample:2 pwm:1.0000 asymmetry:-0.6716 tolerance:n/a
01:49 sample:1 pwm:1.0000 asymmetry:5.4062 tolerance:n/a
01:46 PID_CALIBRATE HEATER=extruder TARGET=240 TOLERANCE=0.01 WRITE_FILE=1
```
My concerns here are:
- The calibration "completes" suspiciously quickly -- samples are being reported in bursts, with the whole thing completing in a minute or so after reaching temp.
- In addition, oftentimes the asymmetry will bounce between +ve and -ve values, then the tolerance will suddenly slam to 0. _NB: I do recall the tolerance suddenly hitting zero from tuning the bed, but cannot recall the significance or remediation._
- Not shown, but if I reduce the target temp to 220, I'll get a "extruder not heating at expected rate" error. I've also tried with increasing the tolerance to 0.02.

I'm unsure if this is a bug (especially the tolerance suddenly hitting a flat zero) or expected behaviour that isn't elucidated in the docs.

Here's some supporting info, if it helps:

Screengrab 1:
[image]

Screengrab 2:
[image]

[klippy.log (link to sprunge.us pastebin)](http://sprunge.us/SVHJE6)

Many thanks for any insights anyone could offer!

dans98, 2023-09-28:
@kjkent I'm under the weather at the moment, but can you do another test where you write to the log, and send me that file? 

At first glance it almost seems like the thermistor or heater cartage isn't making consistent contact with the block, as you pwm and asymmetry values are all over the place, and pwm values shouldn't be able to go negative.

kjkent, 2023-09-29:
Hi @dans98, sorry to hear you're not feeling so great; I hope you have a speedy recovery.

Apologies for not including the PID log previously; I just learned today that it's output to `/tmp`. I did five more PID attempts today under different circumstances, the conditions of which and the results are below.

**TL;DR:** Likely not a bug; my heater and/or thermistor are perhaps lemons :(. Runs 3 & 4 are most valid in terms of testing, with 1 perhaps being bugged, and 2 using original Klipper tuning algorithm. Run 5 is a control.

So, following this I'll look for ways to mitigate, perhaps I'll just need to buy new hardware.

**Setup:**
  - All runs were done in Docker containers, with images based on variations of [this Dockerfile](http://ix.io/4HH8/Dockerfile)
  - If not familiar with Dockerfile syntax, it just clones the git repo of the specified Klipper codebase and readies it for running on a linux host

**Run 1: @dans98 code merged into Klipper master**
- Codebase: [my fork](https://github.com/kjkent/klipper-pid-improvements), which solely cherry-picked the pid-tuning commits [DangerKlippers@e5fc1d4](https://github.com/DangerKlippers/danger-klipper/commit/e5fc1d4face1eb5681f626cae53656865e8b1eac) and [dans98@6eed5c0](https://github.com/dans98/danger-klipper/commit/6eed5c0c379167ab959eb56fe7b6cb05ff975f86) and merged them into code from the Klipper master branch
- Result: A few short results before Klipper errored out. A look at klippy.log showed a divide-by-zero error in the PID tuning code. Looks like perhaps I fumbled the merge (if so, I blame it on the ungodly hour in which I attempted it!).
```log
mcu 'mcu': Exception in serial callback
Traceback (most recent call last):
  File "/opt/klipper/klippy/serialhdl.py", line 56, in _bg_thread
    hdl(params)
  File "/opt/klipper/klippy/mcu.py", line 549, in _handle_analog_in_state
    self._callback(last_read_time, last_value)
  File "/opt/klipper/klippy/extras/adc_temperature.py", line 33, in adc_callback
    self.temperature_callback(read_time + SAMPLE_COUNT * SAMPLE_TIME, temp)
  File "/opt/klipper/klippy/extras/heaters.py", line 87, in temperature_callback
    self.control.temperature_update(read_time, temp, self.target_temp)
  File "/opt/klipper/klippy/extras/pid_calibrate.py", line 141, in temperature_update
    self.set_power()
  File "/opt/klipper/klippy/extras/pid_calibrate.py", line 205, in set_power
    mid = power * ((self.target - peak_low)/(peak_high - peak_low))
                   ~~~~~~~~~~~~~~~~~~~~~~~^^~~~~~~~~~~~~~~~~~~~~~~
ZeroDivisionError: float division by zero
```
- PID plot:
  [image]
- Console output:
```log
20:10 pid_calibrate interrupted
20:10 calibration did not finish in time
20:05 sample:2 pwm:1.0000 asymmetry:7.6393 tolerance:n/a
20:05 sample:2 pwm:1.0000 asymmetry:7.6393 tolerance:n/a
20:05 sample:2 pwm:1.0000 asymmetry:7.6393 tolerance:n/a
20:05 sample:2 pwm:1.0000 asymmetry:7.6393 tolerance:n/a
20:05 sample:2 pwm:1.0000 asymmetry:7.6393 tolerance:n/a
20:05 sample:2 pwm:1.0000 asymmetry:7.6393 tolerance:n/a
20:05 sample:2 pwm:1.0000 asymmetry:7.6393 tolerance:n/a
20:05 sample:1 pwm:1.0000 asymmetry:29.5328 tolerance:n/a
20:03 PID_CALIBRATE HEATER=extruder TARGET=240 TOLERANCE=0.01 WRITE_FILE=1
```
- Log outputs:
  - [PID tuning CSV](http://ix.io/4HGP)
  - [Klippy log](http://ix.io/4HGQ)

**Run 2: danger-klipper:master**
- Codebase: danger-klipper master branch
- Result: PID tuning `completed`
- PID plot:
   [image]
- Log outputs:
  - [PID tuning CSV (merge of data from heatttest.txt)](http://ix.io/4HHp)
  - [Klippy log](http://ix.io/4HGS)

**Run 3: dans98:pid_calibrate**
- Codebase: dans98's fork, pid_calibrate branch (currently https://github.com/DangerKlippers/danger-klipper/pull/62)
- Result: Klippy error `heater "extruder" not heating at expected rate`
- PID plot:
  [image]
- Log outputs:
  - [PID tuning CSV](http://ix.io/4HGU)
  - [Klippy log](http://ix.io/4HGV)

**Run 4: dans98:pid_calibrate, with heater_verify conditions relaxed**
- Codebase: As above, but this time with the following in `printer.cfg` to allow measurements to stray further before erroring out:
```TOML
[verify_heater extruder]
max_error = 250
check_gain_time = 60
hysteresis = 7
heating_gain = 1
```
- Result: As above (Klippy error regarding heating rate)
- PID plot:
  [image]
- Console output:
```log
22:43 Internal error on command:"PID_CALIBRATE"
22:43 Klipper state: Shutdown
22:42 sample:8 pwm:-2.4138 asymmetry:-1.7001 tolerance:3.7209
22:42 sample:7 pwm:-2.7209 asymmetry:-0.4075 tolerance:3.7209
22:42 sample:6 pwm:1.0000 asymmetry:8.8133 tolerance:2.4750
22:42 sample:5 pwm:-1.4750 asymmetry:10.6939 tolerance:2.4750
22:42 sample:4 pwm:1.0000 asymmetry:4.4502 tolerance:2.1204
22:42 sample:3 pwm:-1.1204 asymmetry:-9.4493 tolerance:n/a
22:42 sample:2 pwm:1.0000 asymmetry:-5.1476 tolerance:n/a
22:42 sample:1 pwm:1.0000 asymmetry:21.4637 tolerance:n/a
22:40 PID_CALIBRATE HEATER=extruder TARGET=240 TOLERANCE=0.01 WRITE_FILE=1
```
- Log outputs:
  - [PID tuning CSV](http://ix.io/4HGW)
  - [Klippy log](http://ix.io/4HGX)

**Run 5: Bed tuning as control**
- Codebase: As above (dans98:pid_caibrate)
- Result: PID tuning completed successfully
- PID plot:
 [image]
- Log outputs:
  - [PID tuning CSV](http://ix.io/4HH5)

dans98, 2023-09-29:
@kjkent having looked at your raw data I think you have something weird going on with your thermistor.

The following happens (at a high level) when you run pid calibration. 

1.  The target temp  is windowed. The lower bound is 2.5 degrees below the target temp, and the upper bound is 2.5 degrees above it.
2. When the thermistor is below the lower bound  the heater is turned on, until the temperature rises above the upper bound. When the temperature rises above the upper bound, the heater is turned of, until the temperature drops below the lower bound.
3. The heater output is recalculated after each interval, and step 2 repeats until iterations max out, or the tolerance is meet.

I've quickly looked at your raw data, and your machine is reporting huge temperature shifts from one reading to the next when it comes up to temp. For example on your 1st test I see 31 times where it reported a temperature change of over 20 degrees from reading to the next (0.3 seconds). Unless you are hitting your hotend with a blow torch or liquid nitrogen that really shouldn't happen. 

Those rapid temperature changes are throwing the algorithm (new and old) all out of wack.

kjkent, 2023-09-29:
Thanks for your insights, @dans98 ! I hadn't spotted that and likely wouldn't have. 

This (volcano clone) hotend replacement has a long vertical cutout for a new-style long thermistor, but will likely fit the stubby one I had in my old heatblock. I'll swap them over and see what happens! I'll report back shortly 👍.

kjkent, 2023-10-18:
Hey @dans98 , just chiming in - apologies for the radio silence, had a lot going on and was working through the tuning issues, finally sorted it, I think, and will write up the fairly interesting (imo) conclusion later this afternoon. 

I was also delayed for a few days debugging why the MCU had apparently just dropped dead. Turns out I had accidentally connected something to the reset pin 😎

rogerlz, 2024-02-18:
@kjkent are we ready to close this one?

rogerlz, 2024-10-23:
closing this for now.. feel free to reopen if needed

closed 2024-10-23 by rogerlz