KalicoCrew/kalico #76: Feature request: allow hardcoded CANBUS UUIDs

bryanboettcher, 2023-10-27:
## Use-case:
I have several nearly identical printers in a print farm.  They all have custom toolheads, with CANBUS toolboards -- but all the toolheads are built identically to serve as warm-swap assemblies immediately after a failure.  The expectation is regardless of the failure, I can swap a new toolhead in, restart the print, and then diagnose and repair the swapped toolhead at a better time.

## Problem
The current state of the Klipper CANBUS implementation requires a step to query the bus and identify all UUIDs that are returned, and update a configuration file with the correct UUID.  My understanding is that this UUID is read from either a chip identifier, or a hardcoded section of ROM, and is not changable by the end user.

## Proposed Fix
What would be changable by the end user though, is a constant value set by the Makefile.  If an option (under "Enable low-level options", requiring an extra step to access) was added to allow the user to specify a UUID, I would be able to flash all my toolheads with the same random ID.  Since my machines will only ever run one toolhead at a time, this would allow me to swap the toolhead and restart the print, without having to open a terminal and query for CAN hardware, then update configuration files.  

## Mansplaining why this is an easy fix despite no experience in this codebase
My expectation is this hardcoded value could be stored in a compile-time array while the "reply to UUID request" code can check this array for a value, and if absent populate it from the original UUID mechanism (then always return the array contents when queried).  Setting the option would pre-populate the array, so the code at runtime would not know (or care) if the array had been filled via compile-time directive, or queried earlier.

jtrmal, 2023-10-27:
Honestly this sounds reasonable, IMO 🤷

rogerlz, 2024-02-17:
Allowing a custom string would be too much of a change, potentially breaking Katapult scripts too.
The reason is that Klipper doesn't really use the raw uuid from the chips, but a fast-hashed and truncated to 48-bit version of it. 

I have put together a simple change to allow the user to specify the hash string instead of using the raw uuid by default
https://github.com/DangerKlippers/danger-klipper/pull/156

I was able to compile and flash a board with this change, but I don't have >1 can boards installed at the moment to confirm that the same can_uuid shows up.

rogerlz, 2024-02-22:
This has been tested by a member of our discord community.

closed 2024-02-22 by rogerlz