KalicoCrew/kalico #77: Feature Request: Add beacon Stock Support

Kizime123, 2023-10-27:
Use-case:
Allow Beacon to be used with default Dangerklipper

Problem:
Klippers/ Kevin most recent pull request had made clear Beacon may need a new home. Dangerklipper is the fork to do such.

Proposed Fix:
Add numpy for downloaders, then include beacons code.

kmobs, 2023-10-28:
Care should be taken in order to prevent having versioning issues. 

I've seen the idea of submodules floating around. 

I think maybe it's time that a formal plugin system be implemented. Kevin has also mentioned he doesn't like the extras folder being used in that manner, but I feel a strong plugin system might prove very useful for all the various additions that don't get merged. 

Current Moonraker is handling the plugin system which isn't necessarily an issue either.

YanceyA, 2023-10-31:
Kevin mentioned a previous plugin discussion on plug-ins in the G10 firmware retraction PR for mainline. Here is a link to the forum post in klipper. https://klipper.discourse.group/t/possible-klipper-plugins-instead-of-macros/9819

rogerlz, 2023-11-02:
numpy is already in the [klippy-requirements](https://github.com/DangerKlippers/danger-klipper/blob/master/scripts/klippy-requirements.txt#L11) so not sure whats missing about this..

I agree with @kmobs about the versioning concern, and we are still trying to figure out the best approach. It would be nice if beacon developers would PR the code to DK directly, but that would make it challenging for them to keep it updated in both places, as it would not be nice to force users to use DK to use beacon.
We thought about submodules, but they are a pain to maintain.

By the way, DK now has a [plugins](https://github.com/DangerKlippers/danger-klipper/pull/82) system that allows you to add modules without making your git repo 'dirty'.

Kizime123, 2023-11-02:
> By the way, DK now has a [plugins](https://github.com/DangerKlippers/danger-klipper/pull/82) system that allows you to add modules without making your git repo 'dirty'.>
Perfect! Great addition, are you guys thinking of having the plugins posted in a plugins repo where can be easily accessed/ see what are the plugins? Should be great for updating stuff however!

rogerlz, 2023-11-14:
@Kizime123 For now, there are no plugins, but we have been discussing something along those lines.. probably a repository dedicated to listing known plugins so users can find them easily. 

I don't think we will be adding plugins directly to this repo, as the main goal is only to merge new features that affect the main codebase (and can't be a plugin) and features that have been PR to mainstream but got denied.

I will close this one, but feel free to ping me on discord if you want to chat more 👍