KalicoCrew/kalico #85: Good Sensorless axis is homed twice if homing_positive_dir is False

nielsvz, 2023-11-04:
Using the new sensorless feature I noticed that the Y axis of my K3 is homed twice, likely because homing_positive_dir is False and the distance moved is negative.

### Original code - `homing.py`:
https://github.com/DangerKlippers/danger-klipper/blob/5cccd3b658e776924293ae4ea7b3c8bcf712fe6f/klippy/extras/homing.py#L285-L288

Adding some additional logging to the code:
```
# X
homing_axis_distances: [69.30625]
hi.retract_dist: 40.0
needs rehome: False
# Y
homing_axis_distances: [-94.00625000000001]
hi.retract_dist: 40.0
needs rehome: True
# Z
homing_axis_distances: [-8.01625]
hi.retract_dist: 0.0
```

If I convert the `dist` variable to an absolute value it works as expected (axis is homed only once):
```python
if any([abs(dist) < hi.retract_dist for dist in homing_axis_distances]): 
``` 

Homing the axis with the absolute value:
```
# X
homing_axis_distances: [69.30625]
hi.retract_dist: 40.0
needs rehome: False
# Y
homing_axis_distances: [-92.80312500000001]
hi.retract_dist: 40.0
needs rehome: False
# Z
homing_axis_distances: [-8.03]
hi.retract_dist: 0.0
```

lraithel15133, 2023-11-04:
I can confirm this behavior on my K3 as well

bwnance, 2023-11-06:
ty for this! included in #90

rogerlz, 2023-12-19:
https://github.com/DangerKlippers/danger-klipper/pull/90 is merged now.. thanks!

closed 2023-12-19 by rogerlz