KalicoCrew/kalico #86: Feature Request: Add a separate variable for the back-off distance after sensorless homing

nielsvz, 2023-11-04:
Sensorless homing is a bit tricky with high current low impedance motors (LDO 2804AH). I find they're most reliable when using the following steps, per axis:
1. Lower the run_current to the home_current
2. Home the axis
3. Back off 5mm
4. Raise the run_current back to the configured value
5. Back off the axis to a minimum of 0.5-1.0x the rotation distance (20-40mm in my case)

It would be nice if the good sensorless feature had an additional setting, such as `sensorless_minimum_dist`, that value could be compared to the distance moved as well (currently using `homing_retract_dist`).

Homing would then follow this process, per axis:
1. Lower the run_current to the home_current
2. Home the axis
3. Back off by `homing_retract_dist`
4. Raise the run_current back to the configured value
5. Back off the axis to the endstop position minus `sensorless_minimum_dist`
6. Check if the distance moved is at least `sensorless_minimum_dist` and re-home if required

numanair, 2023-11-04:
Duplicate of #86?

nielsvz, 2023-11-04:
I believe the difference between this and #78 (I think you meant that issue?) is that this request doesn't change the homing process, but it would add a second variable to distinguish between the backoff distance (homing retract) and the minimum distance required to reliably home with sensorless.

numanair, 2023-11-04:
Yep, mixed up my tabs, thanks!

It sounds like they are achieving the same goal, but I like the additional flexibility of this one.

lraithel15133, 2023-11-04:
> I believe the difference between this and #78 (I think you meant that issue?) is that this request doesn't change the homing process, but it would add a second variable to distinguish between the backoff distance (homing retract) and the minimum distance required to reliably home with sensorless.

#78 is mostly asking for built-in checks for sensorless homing failures. Seems you just want a separate configurable variable for minimum homing distance. I think this would be an easy change. Make it default to retract distance, but make it configurable

nielsvz, 2023-11-04:
> #78 is mostly asking for built-in checks for sensorless homing failures. Seems you just want a separate configurable variable for minimum homing distance. I think this would be an easy change. Make it default to retract distance, but make it configurable

Correct, without a minimum distance I find sometimes it doesn't trigger at all.

bwnance, 2023-11-06:
@nielsvz is the problem that if you set the retract distance to 20-40, it backs off too much after a successful home? because other than that, it should result in the behavior you want

bwnance, 2023-11-06:
#90 will resolve this

rogerlz, 2023-12-19:
https://github.com/DangerKlippers/danger-klipper/pull/90 has been merged now

closed 2023-12-19 by rogerlz