KalicoCrew/kalico #87: Feature request: Improve the way saved variables are managed

fbeauKmi, 2023-11-04:
Currently, saved variables are stored at the end of the printer.cfg file in an odd structure. In my humble opinion, there are several areas for improvement.

- Store these variables in a separate file from printer.cfg (without those ugly comments structure). Hopefully, this will simplify the structure. This file would simply be read after printer.cfg when loading the configuration.
- Allow values to be overridden (as is alreadyin the rest of the configuration). Indeed, in the case of a distributed configuration, it is currently impossible to perform a SAVE_CONFIG, klipper returns an error, (I don't know the exact mechanism). -
- So, commenting on configuration variables becomes obsolete

I hope I've made myself clear, thanks for your work!

Piezoid, 2023-11-04:
> * Allow values to be overridden (as is alreadyin the rest of the configuration). Indeed, in the case of a distributed configuration, it is currently impossible to perform a SAVE_CONFIG, klipper returns an error, (I don't know the exact mechanism).

In the current state the saved values override the user values, if we simply remove the check.
That's what you want, right?

Maybe we can turn the error into a warning? 

1. Ideally the user should be warned when a overridden value is changed in the manually managed config: "The change to the parameter x has no effect as it is overridden by a saved value". However, this require tracking config changes,
2. Warn after `SAVE_CONFIG`: "You might want to remove parameter x which is now overridden by a saved value",
3. Warn every klippy restart,
4. `2.` but also logs overridden parameters at boot. This could be generalized to duplicate parameters across includes.

fbeauKmi, 2023-11-05:
> In the current state, the saved values override the user values, if we simply remove the check. That's what you want, right?
Yes, it is.
 
> Maybe we can turn the error into a warning?
>
>     1. Ideally the user should be warned when a overridden value is changed in the manually managed config: "The change to the parameter x has no effect [...]

TBH, i didn't think about warnings.
Is a setting as `allow_saveconfig_to_override : true` is possible/good idea ? with default value = false. When false you got the actual behavior, when true, no really need to warn as you know what you do.

MasturMynd, 2024-02-27:
> * Allow values to be overridden (as is alreadyin the rest of the configuration). Indeed, in the case of a distributed configuration, it is currently impossible to perform a SAVE_CONFIG, klipper returns an error, (I don't know the exact mechanism). -

Wanted to mention that this portion of the inquiry has been resolved via #153

rogerlz, 2026-10-21:
The main blocker here, SAVE_CONFIG failing with a distributed config, was solved by #153: with `autosave_includes: True` in `[danger_options]`, SAVE_CONFIG updates the included files and backs them up. Moving the autosave block to a separate file isn't planned for now. Closing, feel free to reopen if that part still matters to you.

closed 2026-10-21 by rogerlz