KalicoCrew/kalico #102: Feature Request: Add support for PrusaSlicer's binary g-code standard

VinnyCordeiro, 2023-12-03:
On [Klipper's Discourse](https://klipper.discourse.group/t/feature-request-binary-g-code-bg-code-support/11803) that feature was requested as well, but the tone of the dialogue there indicates that it has a very slim chance of being implemented.

Prusa made available a library for the conversion from text to binary and from binary to text through the AGPL v3 license: https://github.com/prusa3d/libbgcode

Argument for this implementation:

* it dramatically reduce the file size of the g-code, making it possible to be transmitted faster while also occupying less space on the target printer.

Arguments against this implementation:

* only supported by PrusaSlicer right now;
  * counter-argument: this may not be the case in the near future, as popular forks like SuperSlicer and OrcaSlicer may implement this feature as well.
* make it impossible to read the g-code file to find possible errors;
  * counter-argument: humans reading g-code isn't a concern of Klipper, that should be handled by a GUI like Mainsail, Fluidd, or even OctoPrint.
* may need to implent some sort of error checking to ensure there wasn't any corruption of the file while being transmitted/stored.

Does anyone have another point to add to the discussion? I myself think this isn't absolutely necessary, more like a "nice to have" feature.

kmobs, 2023-12-03:
I agree that it's not necessary, and if anything I think that it's additional code burden to maintain with little to no benefit in our class of devices. I understand the utility in slower networked devices with smaller storage capacities like the Prusa machines.

In the setting of nearly everyone running a Pi or better in the klipper/dk world, I don't think it really has much in the way of utility.

dalegaard, 2023-12-04:
Hi,

I think this can be done in `virtual_sdcard` without much extra maintenance burden. To me it seems like a useful feature to get printing a bit faster.

Best regards,
Lasse

blownupp, 2023-12-09:
While I think it would be a neat option, it only takes 2-3 seconds to send a 50mb GCODE file to my printer, and it’s not like I sit there to wait for it to transfer anyway.

bryanboettcher, 2023-12-27:
@VinnyCordeiro would this be solvable for you with a post-processing script?  that seems quite doable.

VinnyCordeiro, 2023-12-31:
> @VinnyCordeiro would this be solvable for you with a post-processing script? that seems quite doable.

That could work.

closed 2024-01-20 by VinnyCordeiro