KalicoCrew/kalico #111: Feature request: Better support for PTC chamber heaters

TheRealDeathsneeze, 2023-12-20:
Currently, one of the more effective methods to control a chamber heater is done in a somewhat hacky way by configuring the PTC heater as a reverse temperature fan along with a temperature fan to circulate hot air. An example configuration is shown:

```
[temperature_fan chamber_heater]
sensor_type: Generic 3950
sensor_pin: #chamber_thermistor
min_temp: 0
max_temp: 120
gcode_id: chamber_heater
pin: #to chamber heater ssr
target_temp: 0
shutdown_speed: 0
control: pid
pid_Kp: 2
pid_Ki: 5
pid_Kd: 0.5
reverse:true

[temperature_fan chamber_heater_core]
sensor_type: Generic 3950
min_temp: 10
max_temp: 250
sensor_pin: #ptc_core_thermistor
gcode_id: chamber_heater_core
pin: #to large fan
target_temp: 65
shutdown_speed: 1
control: pid
pid_Kp: 200
pid_Ki: 15
pid_Kd: 400
kick_start_time: 0.5 

```
This results in a behavior where the chamber heater is turned on by setting the "temperature fan" called chamber_heater to the desired chamber temperature, at which point the chamber heater's will will enable as soon as the temperature of the core reaches the defined target_temp, in this case 65 degrees. 

Instead, it would be simpler and more consistent with the other heaters used typically in a printer if Danger Klipper would have a built in heater_chamber heater type. Then a config could be defined in a more consistent manner, perhaps in a similar fashion to how the [extruder] section of configuration is set up.  Then a minimum config section could be defined specifying the pins for the heater_core, core_thermister, chamber_thermister, and chamber_fan as well as their respective minimum and maximum temperatures.

lraithel15133, 2024-02-03:
Documenting here as a solution using currently existing modules (less hacky than the config block in the OP, which I also made)

`[heater_generic ch]
sensor_type: NTC 100K MGB18-104F39050L32
sensor_pin: PF5
#gcode_id: C
heater_pin: PD14
min_temp: 0
max_temp: 130
control: pid
pid_Kp: 70.184
pid_Ki: 0.376
pid_Kd: 3274.071

[controller_fan ch_fan]
pin: PD15
heater: ch
idle_timeout: 60`

TheRealDeathsneeze, 2024-02-07:
oh I didnt realize that was your config block! I do like the new suggested config as well, although it does differ slightly in function, but it does at least benefit from klipper seeing it as a heater and thus applying heater verification checks

lraithel15133, 2024-02-07:
> oh I didnt realize that was your config block! I do like the new suggested config as well, although it does differ slightly in function, but it does at least benefit from klipper seeing it as a heater and thus applying heater verification checks

in https://github.com/DangerKlippers/danger-klipper/tree/chamber_heater we're working on it, its not currently functioning but its getting there. Setting up the ability to essentially combine config blocks. In this example, we're combining controller_fan and heater_generic

rogerlz, 2025-01-21:
This is being discussed over discord now: https://discord.com/channels/1297243471442214913/1328796693734494238

closed 2025-01-21 by rogerlz