KalicoCrew/kalico #112: Feature Request: Allow automatic detection of build envelope for XY

bryanboettcher, 2023-12-20:
During machine configuration, it is necessary to establish the bounds the toolhead can safely move.  

If the homing routine for an axis was expanded to display the distance travelled, this could dramatically simplify the initial setup of a machine.  A procedure where the end user manually positions the toolhead to a known position (such as axis minimum) and then homing will get what the machine thinks the travel on that axis is.  Knowing, for instance, that my X axis has 304mm of travel (as output by a DEBUG statement) will allow me to configure position_min and position_max to correctly position 0,0 on my print surface while still respecting the physical limitations of the machine.

Outputting the distance travelled during homing will remove the very manual process of moving the toolhead a fraction of a millimeter at a time while checking for a slip of paper being snagged by the toolhead against the ends of the machine.

rogerlz, 2023-12-21:
loved the idea

GiulianoM, 2024-02-18:
TBF, I kind of like the idea of using Sensorless to find both Max and Min positions, simply by smacking into the ends like you would with homing.

MartijnGevaert, 2024-04-24:
Being able to measure the total travel also has 2 other potential features, which are possible if you manually set an expected value for each axis.

- **Knowing for sure sensorless homing is successful**
If the total distance doesn't match up to a pre-configured value, you can detect a homing failure
- **Detecting gantry de-racking**
On some gantry's, such as CoreXY, the gantry can become unracked over time due to, for example, belt tension becoming uneven. An unracked gantry will reduce the total amount of travel available on an axis. This means that measuring the total travel distance and comparing it with the expected value can also be used to detect an unracked gantry.

These features are different from the original feature requested here, but I assume that adding in the features I proposed would affect how the original feature on this ticket would be implemented also.