KalicoCrew/kalico #114: Feature request: add danger options flag to ignore min/max temps for thermistors

TheRealDeathsneeze, 2024-01-01:
During initial configuration one sometimes runs into issues where it would be desirable to disable the min/max temperature checks so that one could continue testing/troubleshooting during initial setup. 

Some safeguards could additionally be put into place to safeguard against this potentially dangerous configuration. For example, the ability to start a print or to turn on heaters(possibly allowing heaters to run for a very brief period to verify wiring) could be disabled while this option flag is enabled.

conlank, 2024-01-01:
Another way a similar outcome could be achieved (ease of setup) would be to allow the MCU with the ADC error to name the pin/thermistor that called the shut down.

TheRealDeathsneeze, 2024-01-01:
> Another way a similar outcome could be achieved (ease of setup) would be to allow the MCU with the ADC error to name the pin/thermistor that called the shut down.

that would honestly be really amazing! It's a bit annoying figuring out exactly which one it is that is erroring out on you when there are several of them(which is honestly the biggest reason for me to make this feature request)

rogerlz, 2024-01-16:
You can use `adc_ignore_limits` now. There are no safeguards for now, so you can set your printer on fire if you are not careful. :)

I still want to implement @conlank's idea someday, but modifying the shutdown command output is complex. I have added to my TODO list

closed 2024-01-16 by rogerlz