KalicoCrew/kalico #126: Feature request: Expand Klipper make menuconfig options to disable optional/unneeded features

VinnyCordeiro, 2024-01-15:
Since version 0.10 Klipper doesn't fit on the less powerful AVR microcontrollers anymore, namely:

- atmega168
- atmega328
- atmega328p
- atmega32u4

In 2022 I made some experiments to try and solve that, like:

- manually compiling avr-gcc from gcc 12.1.0 (I followed the instructions on this blog post: https://blog.zakkemble.net/avr-gcc-builds/), since the avr-gcc that is packaged on Debian is version 5.4, which means we don't have access to a lot of optimization options and flags;
- also manually compiled and installed version 2.1.0 of AVR-LibC.

After these were installed, proceeded to compile a lot of tests, including manual Makefile edits, and while using optimization flags/option did indeed reduced the compiled firmware, they were still too big to fit an ATmega32U4 mcu (which was my main concern at the time).

What really did the trick was manually commenting out some files on the `/src` directory. On my case, I have designed a toolhead control keyboard, so I could use it to move the toolhead without the need to go to my computer for that. Because of the use case, I commented some files entirely, like `lcd_hd44780.c`, `lcd_st7920.c`, `neopixel.c`, `sensor_adxl345.c`, and `thermocouple.c`, which made the resulting firmware finally fitting the 32u4.

You already select features to be disabled on the `make menuconfig` environment, as seen here:

[image]

and here:

[image]

, but they are only available for the atmega168, atmega328, and atmega328p microcontrollers, leaving the 32u4 uncovered, and even so they are not enough to make the firmware fit as is. So, my request is to expand the features that can be removed, so you can make Klipper fit those AVR mcus again.

VinnyCordeiro, 2024-01-16:
> @VinnyCordeiro would you please test this #127? all the `*.c` files you mentioned are already covered, but atmega32u4 was not enabled to use the feature filter

With that change the code now compiles and fit the 32u4 mcu, but is still too big for the others.

rogerlz, 2024-01-16:
thanks!

I have added options to disable I2C and SPI, which makes it compile for atmega328. 

I was only able to compile atmega168 after removing pwm and adc, which would probably make it useless.. but let me know your thoughts.

Piezoid, 2024-01-16:
I did this a while ago to optimize code size: https://github.com/Piezoid/klipper/commits/detail_kconf/
It adds menuconfig option for optional features, but I only added link filtering to stm32.
There is also a commit that enables LTO, although not all compilers support it and it's currently not gated by a feature flag.

rogerlz, 2024-01-16:
Thanks both! I have covered I2C and SPI with my last commit. I will pull some more from @Piezoid's branch and see if I can still get atmega168 compiled, otherwise I will drop the support.

ganzgustav22, 2024-01-18:
Hello,

It looks like there hasn't been any recent updates on this
DangerKlipper github issue during the last two hours.
If you created this issue and no longer consider it open, 
then please login to github and close the issue. Otherwise,
if there is no further activity on this thread then it will be
automatically closed in a few minutes.

Best regards,

~ Your annoying GitIssueBot

PS: I'm just an automated script, not a human being.

VinnyCordeiro, 2024-12-05:
Since now DK is now Kalico I can't find which branch have this feature anymore, so I'm reopening this issue.

rogerlz, 2025-02-01:
I have merged https://github.com/KalicoCrew/kalico/pull/560, and I tested with 328 with adc selected, and it fits, making it useful. I will probably hide atmega168 from the menuconfig soon.

closed 2025-02-01 by rogerlz