KalicoCrew/kalico #147: stm32h7.c hardcodes

calithameridi, 2024-02-07:
#

calithameridi, 2024-02-07:
I find that bodging the value of pll_base to 4 MHz allows me to successfully build an image with 400 MHz clock. Obviously this is not a permanent fix although it may suffice in providing pre-built firmware images to Corevus beta-testers

[image]
[image]

Fisheiyy, 2024-02-08:
This would also be good to PR into mainline klipper and fix it for EVERYONE, not just DK

closed 2024-02-09 by bwnance

calithameridi, 2024-02-09:
> This would also be good to PR into mainline klipper and fix it for EVERYONE, not just DK

yeah but they don't have a convenient way of reporting issues