KalicoCrew/kalico #149: Fix SAVE_CONFIG to apply to imported .cfgs

lraithel15133, 2024-02-08:
Currently, if you have an imported cfg that save_config wants to override, it cannot comment out the lines in an imported cfg. configfile.py does properly override and can be overwritten, but the initial commenting out in an imported cfg cannot be performed.

With extruder block in a separate config, no PID settings in the save_config variable block, and no extruder section in printer.cfg, when you hit save_config, you get this error
"SAVE_CONFIG section 'extruder' option 'control' conflicts with included value"
However, if you add something to the variable block on printer.cfg, it is able to override those values which do properly apply.

 Proposal is just to fix save_config to also check any included files for the particular block it is looking for (in this case, [extruder])

MasturMynd, 2024-02-08:
During our discussion, I proposed something similar to the following. The only missing bit of information would be the path of the included file

```
def _read_config_file(self, filename):
	try:
		f = open(filename, 'r')
		data = f.read()
		f.close()
	except:
		msg = "Unable to open config file %s" % (filename,)
		logging.exception(msg)
		raise error(msg)

	## store the current file's data in memory for safe keeping
	output = data.replace('\r\n', '\n')

	## define a regexp that will match any include block
	regex = re.compile('[include(.*)]')

	## iterate through the current config
	for line in output.split('\n'):

		## identify include blocks
		if regex.search(line):
		
			## pull the path from the include block and trim any superfluous whitespace
			include_name = ## pull the file path from the line .trim()
				
			## recursive call to ensure that all potential locations are checked
			sub_data = _read_config_file(self, include_name)
				
			## format new lines properly
			sub_data = sub_data.replace('\r\n', '\n')
				
			## replace the include block with the contents of the config file
			output = output.replace(sub_data, line)
		
	## return the completed output
	return output
```

lraithel15133, 2024-02-17:
Merged in #153

closed 2024-02-17 by lraithel15133