KalicoCrew/kalico #172: Feature Request: New Klippy.log file for each SBC restart

Kizime123, 2024-03-11:
I have a 50mb klippy.log file currently, its a pita to upload to discord with troubleshooting. 
Whenever people run into issues its normally something that only is needed to be seen in the most recent SBC restart. Having the log file restarted on each SBC wakeup would not take 50mb and be annoying (:

lraithel15133, 2024-03-11:
I think this is a good danger_option, probably

rogerlz, 2024-03-12:
Logging is configured before the config is read, so I can't make it a danger_option.
I can make it a CLI argument, though. It would require modifying the `klipper.service` file.

here is a draft diff
```diif
diff --git a/klippy/klippy.py b/klippy/klippy.py
index 3bee3b01..c1d39839 100644
--- a/klippy/klippy.py
+++ b/klippy/klippy.py
@@ -306,6 +306,7 @@ class Printer:

     def run(self):
         systime = time.time()
+        self.bglogger.doRollover()
         monotime = self.reactor.monotonic()
         logging.info(
             "Start printer at %s (%.1f %.1f)",
diff --git a/klippy/queuelogger.py b/klippy/queuelogger.py
index 8b980238..c38faa51 100644
--- a/klippy/queuelogger.py
+++ b/klippy/queuelogger.py
@@ -26,7 +26,7 @@ class QueueHandler(logging.Handler):
 class QueueListener(logging.handlers.TimedRotatingFileHandler):
     def __init__(self, filename):
         logging.handlers.TimedRotatingFileHandler.__init__(
-            self, filename, when="midnight", backupCount=5
+            self, filename, when="S", interval=24*60*60, backupCount=5
         )
         self.bg_queue = queue.Queue()
         self.bg_thread = threading.Thread(target=self._bg_thread)
```

rogerlz, 2024-03-29:
https://github.com/DangerKlippers/danger-klipper/pull/181

closed 2024-03-30 by rogerlz