KalicoCrew/kalico #198: Feature Request: Native support for heated part cooling control (fan+heater+temp sensor)

dlcklazer, 2024-04-06:
Introduced in Discord 4/5/24. [Link](https://discord.com/channels/1029426383614648421/1164249107938947084/1225440911337590815)
### Abstract
Adding an air heater to the part cooling duct, to heat the cooling air above chamber temp, resolves many print quality, isotropy, and strength issues with high temp filaments that are prone to warp. In "high temp" printing it is generally assumed one "cannot use cooling", but this is not true if the cooling air is sufficiently preheated. In fact, this method becomes compulsory to approach true strength potential with these feedstocks in non-commercial, high-end printers.
Klipper (and slicers I know of) have no native means to assign n+, independent heaters/sensors to a toolhead, nor control it/them in parallel (or in the loop) with the native part cooling fan controls that Klipper & Slicers definitely do control now.

This is a separate dynamic from chamber heat control, but dumping this much extra heat into the chamber could present challenges that are at odds with other workflows that try to keep chamber temperature up, down, or steady. In my case I need to actively cool my chamber, which I do now by propping the door open. That's another issue to resolve later, with other controls that are preferably in the loop too.

### My Awareness
I'm really new to DK, so doing my best not to suggest things that exist. I've perused some related functionality that blends some of the heater and fan modules to overcome overlap between existing fan & heater modules that can't be used together sensibly:
#168 Fan Mixer - this could be useful if two airflow sources are needed for the heated PC.
#194 Controller temp fan - two competing sources of fan input is maybe the problem space, but with heater controls in the loop too.
#129 ADC Ignore - this might help with some of the problems by chamber temp drift from extreme heated part cooling thermal loads.
### Requirements for my build

**Safety**

- when part cooling air heater is on, the PC fan must be on.
- when PC air heater is commanded OFF, the part cooler fan should be ON for a user-configurable post flow to prevent latent heat roasting anything it shouldn't. Some applications won't need it (e.g. metal heater housings). Pumps will cease flow promptly, fans may not.
- when `fan speed` < 100% the heater should be able to maintain target temp without error. Slicers can set PC fan speed very low.
- there should be some awareness between extruder temp and PC air/heater temp to prevent accidental overshoot. 250° C part cooling air could melt your print if you fat finger a temp control override in Mainsail et al.

**Function**

- `M106` commands from slicer to GCODE should be able to control fan speed like a normal printer.
- Ideally the slicer could do the work of picking heated cooling target temperature, which is largely a function of feedstock Tg, but can only really be determined through experimentation.
- - If not possible, there has to be some compulsory prompt in Octo/Mainsail when starting a print that makes you pick a temperature.
- Being able to optimize both air speed and temperature is not something slicers seem capable of doing, but rheology in this application suggests there is and should be a relationship.
- Non-critical temperature fluctuations should not kill the print, but there still need to be ways to cascade higher severity measures as more unsatisfactory conditions are detected. If part cooling temperatures can't be controlled, it would still be preferable to continue the print if everything else is nominal. It may even be OK to keep pumping chamber-temp cooling air (but not ambient temp, which depends on your airflow source).
- There may be potential for in-situ annealing with really hot part cooling air, which is uncharted but exciting territory. For instance, deposit a layer, then run a slow annealing air pass along a path. I know, this is getting complicated.

I have other functions that I can't recall at the moment, but would like to submit this and then add to it as I remember.

TheRealDeathsneeze, 2024-04-06:
that is very well said and a great writeup of the discussion yesterday! also tangentally related:
https://github.com/DangerKlippers/danger-klipper/issues/111

rogerlz, 2024-04-11:
just so I don't forget, I have created a module for this and @dlcklazer is testing it
https://gist.github.com/rogerlz/03ae2b7530cb7587cf652ae4de88a4ae

rogerlz, 2024-06-04:
https://github.com/DangerKlippers/danger-klipper/pull/259

closed 2024-06-04 by rogerlz