KalicoCrew/kalico #221: Feature Request: Add accel_per_hz to csv and png generation

lraithel15133, 2024-04-27:
I tried to do this but I'm too smooth. Basically, I want to pass accel_per_hz into the csv, and then also use it when the graph png is generated. The reason it needs to be passed into the csv is because the current value may not match the value from when the csv was created. 

- Pass it into the generated csv
- Add it somewhere that makes sense on the graph generation (above/below recommended shaper or something)

This should probably also be ported over to bleeding edge.

lhndo, 2024-05-15:
On a similar note, I have this patch that adds the accel per hz parameter (APHZ) to the TEST_RESONANCES command.
I'm too github illiterate to do a PR myself and my coding skills are questionable 😅

https://github.com/lhndo/ResHelper/commit/13cdf5a4a75aab101b79e48178956a58cc65c1a6

rogerlz, 2024-06-12:
Derp's request has been merged https://github.com/DangerKlippers/danger-klipper/pull/224
and LH's suggestion added to the backlog https://github.com/DangerKlippers/danger-klipper/issues/295

closed 2024-06-12 by rogerlz