KalicoCrew/kalico #292: Second extruder motor not spinning when it should on dual extrusion prints

OldGuyMeltsPlastic, 2024-06-10:
I tried installing DangerKlipper on my Tridex using the configuration files from here:
https://github.com/joseph-greiner/tridex_mods/tree/main/printer_configuration
Single extruder prints with either toolhead (let's call them T0 and T1) work as expected.
When I tried a dual extruder print, I noticed that T1's extruder motor was spinning in sync with T0's extruder motor, even though T0 was the "active" toolhead, and the T1 toolhead wasn't moving at all.
Conversely, when T1 is the "active" toolhead, although the toolhead moves through the XY motions of the layer being printed, its extruder motor doesn't spin at all, meaning that no filament is extruded except for a tiny bit of oozing.
I have attached the GCODE file for the print, my printer configuration files for the Tridex, and a link to the video that demonstrates what I am talking about.
[IDEX Issue with DK.zip](https://github.com/user-attachments/files/15778544/IDEX.Issue.with.DK.zip)
https://discord.com/channels/1029426383614648421/1164249107938947084/1249739604358070397

rogerlz, 2024-06-17:
a suggestion has been made via discord: 
https://discord.com/channels/1029426383614648421/1164249107938947084/1250471587392524288

pending feedback

rogerlz, 2024-08-31:
Closing this due to no feedback.. pls reopen if it is still an issue