KalicoCrew/kalico #294: Possible to heat one element with two heaters and two thermistors?

RoboticKoala, 2024-06-12:
Hello! I'm making a custom printer that's extruding some strange material (sorry cannot say much) and I plan to have an extruder with two cartridge heaters and one (or maybe two) thermistors. The cartridge heaters will be the same and I would like to control it with one input, do you think this would be possible? And if I used two thermistors to take the average value from them to determine the temperature? 

I looked on some online forums and they reccomended using danger klipper. Thank you for your help and time, I appreciate it loads and am looking forward to your responses. Thank you again!!

rogerlz, 2024-06-12:
Hey!

DK does not have anything specific to your requirements, but I believe you can already achieve what you need with existing features. 

A few options:
- `[multi_pin]` to control two heaters and one sensor [Config Reference](https://dangerklipper.io/Config_Reference.html#multi_pin)
```
[multi_pin heater]
pins: PB4,PB5

[extruder]
heater_pin: multi_pin:heater
```

- `temperature_combined` sensor to combine and average two temperature sensors [Config Reference](https://dangerklipper.io/Config_Reference.html#combined-temperature-sensor)
```
[temperature_sensor heater_sensor1]
sensor_type: PT1000

[temperature_sensor heater_sensor2]
sensor_type: PT1000

[extruder]
sensor_type: temperature_combined
sensor_list: temperature_sensor heater_sensor1,heater_sensor2
combination_method: mean
maximum_deviation: 20.0
```

- `generic_heater` to have a second heater instance with an independent sensor and pid control [Config Reference](https://dangerklipper.io/Config_Reference.html#heater_generic)
```
[heater_generic extraextruder]
gcode_id: X
heater_pin:
sensor_type:
sensor_pin:
```
and then add `SET_HEATER_TEMPERATURE` to your `PRINT_START` / `PRINT_END` macros.

RoboticKoala, 2024-06-12:
Wow! That's awesome! Thank you for your help, I appreciate it! I'll try it!

rogerlz, 2024-06-12:
Sure! feel free to reopen or ping me on discord if you need help

closed 2024-06-12 by rogerlz