KalicoCrew/kalico #319: BUG: current_change_dwell_time applying to non-sensorless homes

lraithel15133, 2024-06-22:
https://discord.com/channels/1029426383614648421/1164249107938947084/1254144015721300043

Not using sensorless, and not changing current, current change dwell time seems to be affect homing delay. User is using a homing_override, but only variable that has changed between the linked videos is dwell config.

```
[stepper_x]
step_pin = PE2
dir_pin = !PB4
enable_pin = !PC11
microsteps = 32
rotation_distance = 40
endstop_pin = ^EBBCan: PB8
position_min = 0
position_endstop = 300
position_max = 300
homing_speed = 100
homing_retract_dist = 5
homing_positive_dir = true

[tmc2209 stepper_x]
uart_pin = PC10
interpolate = False
run_current = 0.9
sense_resistor = 0.110
stealthchop_threshold = 0
current_change_dwell_time = 0.01

[stepper_y]
step_pin = PF12
dir_pin = !PF11
enable_pin = !PB3
microsteps = 32
rotation_distance = 40
endstop_pin = ^PF4
position_min = 0
position_endstop = 307
position_max = 307
homing_speed = 100
homing_retract_dist = 5
homing_positive_dir = true

[tmc2209 stepper_y]
uart_pin = PF13
interpolate = False
run_current = 0.9
sense_resistor = 0.110
stealthchop_threshold = 0
current_change_dwell_time = 0.01
```

rogerlz, 2024-07-01:
I believe https://github.com/DangerKlippers/danger-klipper/pull/320 fixed this?

bwnance, 2024-07-17:
@lraithel15133 bump

closed 2024-08-31 by rogerlz