KalicoCrew/kalico #321: Feature Request: Multi fan compensation for MPC heating model

OverkillFPV, 2024-06-23:
Like most people, i run a CPAP cooler as well as aux parts cooling and interestingly and i suppose not suprisingly the rscs aux fan provides more cooling than the CPAP. (22.2W for the the CPAP and 26.8W for the RSCS). allowing a list of fans would help improve the effectiveness of the heater  model as far as i understand. This is assuming that i am not suggesting an already possible feature.

YanceyA, 2024-06-23:
How are the fans run during print? 

- Exclusively, one one fan at a time
- In series, both fans run the same % speed always.
- Independently, the fans are operated at different speeds

OverkillFPV, 2024-06-24:
Both fans are run at independent speeds based off the filament settings, the cpap toolhead fan runs dynamically based off overhangs and speeds but the RSCS fan is typically a fixed speed set in the slicer.