KalicoCrew/kalico #322: Dockable probe errors when used with safe_z_home

ressu, 2024-06-23:
I use a Klicky probe on a switchwire. Normally I have a homing override for Z, which works based on the standard klicky principle. Since dockable probe was introduced, there shouldn't be any reason why safe_z_home shouldn't work as well. So when testing it out, the printer seems to home somewhat normally but after returning the probe to the dock returns a `Must home axis first` error.

```
!! Must home axis first: 125.000 100.000 18.255 [0.000]
```

The regular homing override I use and that works as expected is as follows:

```
[homing_override]
axes: z
set_position_z: 0
gcode:
  G90
  G0 Z5 F500
  {% if "xy" not in printer.toolhead.homed_axes %}
    G28 X0 Y0
  {% endif %}
  G0 X125 Y100 F9000
  G28 Z0
  G0 Z5 F500
```

In theory this should be the same as the followsing, which is what I attempted to use:

```
[safe_z_home]
home_xy_position: 125, 100
z_hop: 5
```

Rest of the meaninful portions of the configuration are as follows. I can't immediately spot anything that could cause the conflict in my configuration, so I'm assuming that the dockable probe and safe_z_home currently just aren't compatible with each other.

```
[dockable_probe]
x_offset: 0.0
y_offset: 31

approach_position: 203, 150
dock_position: 203, 150
detach_position: 170, 150
check_open_attach: true
travel_speed: 200
z_hop: 20

speed: 5
lift_speed: 7
samples: 3
samples_result: median
sample_retract_dist: 0.5
samples_tolerance: 0.01
samples_tolerance_retries: 10

drop_first_result: True
dock_retries: 2

[gcode_macro _DockServoMove]
gcode:
  {% set ANGLE  = params.ANGLE|int %}
	
  SET_SERVO SERVO=klicky_servo ANGLE={ANGLE}
  G4 P250
  SET_SERVO SERVO=klicky_servo WIDTH=0.0

[gcode_macro MOVE_TO_APPROACH_PROBE]
rename_existing: MOVE_TO_APPROACH_PROBE_BASE
gcode:
  _DockServoMove ANGLE=135
  MOVE_TO_APPROACH_PROBE_BASE
  _DockServoMove ANGLE=50
  _DockServoMove ANGLE=135

[gcode_macro MOVE_TO_DETACH_PROBE]
rename_existing: MOVE_TO_DETACH_PROBE_BASE
gcode:
  _DockServoMove ANGLE=50
  MOVE_TO_DETACH_PROBE_BASE
  _DockServoMove ANGLE=135
```

ressu, 2024-06-24:
I completely missed the fact that I hadn't updated before running the last tests. With the latest firmware the homing routine no longer errors out, but homing happens at the position where the toolhead is after probe has been attached instead of the designated homing position.

So the routine is as follows, the printer homes X and Y. Toolhead moves to Z homing position, then attaches probe but doesn't return back to the Z homing position. Printer then homes Z and docks the probe.

And since video is worth thousand words: https://youtu.be/7XwtUUALcHM

fbeauKmi, 2024-06-26:
Thanks for this feedback.
Unfortunatly, I wasn't able to reproduce your issue on my setup. Could you precise the DK version and the branch you are running ?
I notice approach_position is equal to dock_position. I suspect this interfere with move_avoiding_dock feature. Could you try to set ``dock_position: 203, 151``  to confirm the issue?

ressu, 2024-06-26:
Turns out you are right, it seems to be the dock avoidance. Moving the dock back by a single millimeter makes homing work as intended.

And just for the record, this is DK v0.12.0-418-g8adfaeb9 on the master branch.

rogerlz, 2024-07-01:
thanks both!

closed 2024-07-01 by rogerlz