KalicoCrew/kalico #346: Add new stuff from dmbutyugin's advanced-features branch

Ataraxia-Mechanica, 2024-08-14:
https://github.com/dmbutyugin/klipper/commits/advanced-features/
Currently bleeding-edge-v2 is a bit old and doesn't support changing PA during printing (master branch of danger klipper has a commit that reverted the change that made printer stutter while changing PA so it isn't affected). One of the new commits from advanced-features also made offset parameter in mm instead of ms which should make it perform even better. There's also a commit that made disabling smoothing extruder base position optional. Bunch of cool stuff in it really.

Ataraxia-Mechanica, 2024-08-14:
Actually no, I am getting reproducible Internal error on command G1 with it :(

Ataraxia-Mechanica, 2024-08-27:
By only adding the [allow dynamic PA commit](https://github.com/dmbutyugin/klipper/commit/9590ffb13d190d52a8e58d083baaa47f869239ca) to BEv2 I haven't encountered any error for a few hours of print time, so I think it can be added.

rogerlz, 2024-09-17:
Thanks for testing what works and what doesn't. Indeed, there are a few changes in the mainline that make advanced features incompatible with dk bleeding-edge-v2. I will look into the dynamic pa commit.

rogerlz, 2026-10-21:
Dynamic pressure advance is now in bleeding-edge-v2 via #1026. Thanks for testing it!

closed 2026-10-21 by rogerlz