KalicoCrew/kalico #388: ability to fully disconnect non-critical MCUs

artemen, 2024-10-16:
I am working in incorporating cartographer into a tool changer, and would like to use only one probe across all of the tools, basically, a 4 pin magnetic connector  on a tool doing 5v,gnd,+can/usb and keep the sensor itself hard fixed on the on the carriage.   would it be possible to prevent klipper from shutting down upon disconnect of the cartographer probe?

rogerlz, 2024-10-16:
yes, if using usb.
add `is_non_critical: true` to your config.

example:
```
[beacon]
is_non_critical: true`
```

artemen, 2024-10-16:
thanks for the response, it looks like CAN MCUs can't be declared non-critical, is there any specific issue preventing this or no one needed it yet? thx!

rogerlz, 2024-10-18:
> thanks for the response, it looks like CAN MCUs can't be declared non-critical, is there any specific issue preventing this or no one needed it yet? thx!

that's correct, CAN mcus are not supported yet. Once mcus are initialized they don't respond to uuid queries anymore, so there is currently no way to discover them. A bit more code is needed and we are not very keen to work on CAN :)

closed 2024-10-18 by rogerlz