KalicoCrew/kalico #394: Extra extruders with tradrack

lukeslaboratory, 2024-10-28:
So, I have a hungry hotend, and a very short path between the hotend & the extruder.
I would really like to be able to use a third extruder in the chain between the tradrack extruder & the main extruder, if nothing else but to do a proper retract so I can use it for endless spool mode.

Right now when filament runs out, its very likely to try to retract the filament out, but the length that it retracts is far less than what is needed to re-feed it into the tradrack feeder extruder. 

Thanks

rogerlz, 2024-12-02:
ping @rsghosh

lukeslaboratory, 2025-01-01:
Hey, @rsghosh can we make this happen sometime? Can very much use this feature daily :)
Luke