KalicoCrew/kalico #398: Feature Request: Automatically shut down hotend fan during input shaper

jaakko000, 2024-10-28:
https://github.com/MRX8024/motors-sync/blob/main/wiki/EN.md 

From above macro consider adding the following portion to [adxl345] or [resonance_tester], wherever it applies. 

```
#head_fan:
#    Toolhead fan, which will be turned off during sync to eliminate noise.
#    This is convenient when the fan has a low temp target and is often
#    turned on, for example in thermal chamber.
```

Why? 
`Do not turn on the hotend heating during synchronization. Working the fan (in general, any fan in the printer) can interfere with correct and more accurate measurement.`

The-Original-Reth, 2024-11-23:
I think it might be better to add this as an option in [reasonance_testeror] as some may prefer to test the effect of the hotend fan and part cooling fan effect upon resonance. 

Something like 
fan_off: fan, part_cooling fan, mcu_fan

This may allow one to decide which fans to turn off and which to keep on

rogerlz, 2024-12-10:
I prefer to add a warning saying that a fan is running when the user starts the testing. Turning the hotend fan off and bypassing the heater control could be dangerous.

closed 2025-03-25 by rogerlz