KalicoCrew/kalico #421: file_browser not working

Matszwe02, 2024-10-31:
I've not been able to make file_browser work. 

## moonraker error
```
Internal error during connect: Choice 'file_browser' for option '{'disabled': <class 'extras.display.menu.MenuDisabled'>, 'command': <class 'extras.display.menu.MenuCommand'>, 'input': <class 'extras.display.menu.MenuInput'>, 'list': <class 'extras.display.menu.MenuList'>, 'vsdlist': <class 'extras.display.menu.MenuVSDList'>}' is not a valid choice

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted
```

## klippy.log

```c
Unhandled exception during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 279, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 214, in _read_config
    self.load_object(config, section_config.get_name(), None)
  File "/home/pi/klipper/klippy/klippy.py", line 198, in load_object
    self.objects[section] = init_func(config.getsection(section))
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/klipper/klippy/extras/display/__init__.py", line 10, in load_config
    return display.load_config(config)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/klipper/klippy/extras/display/display.py", line 327, in load_config
    return PrinterLCD(config)
           ^^^^^^^^^^^^^^^^^^
  File "/home/pi/klipper/klippy/extras/display/display.py", line 220, in __init__
    self.menu = menu.MenuManager(config, self)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/klipper/klippy/extras/display/menu.py", line 748, in __init__
    self.load_menuitems(config)
  File "/home/pi/klipper/klippy/extras/display/menu.py", line 1057, in load_menuitems
    raise error(
extras.display.menu.error: Choice 'file_browser' for option '{'disabled': <class 'extras.display.menu.MenuDisabled'>, 'command': <class 'extras.display.menu.MenuCommand'>, 'input': <class 'extras.display.menu.MenuInput'>, 'list': <class 'extras.display.menu.MenuList'>, 'vsdlist': <class 'extras.display.menu.MenuVSDList'>}' is not a valid choice
webhooks client 548427975376: New connection
webhooks client 548427975376: Client info {'program': 'Moonraker', 'version': 'v0.9.3-1-g4e00a07'}
```

## display.cfg
```yaml
[menu sdcard]
type: file_browser
name:
sort_by: last_modified
enable: {('virtual_sdcard' in printer)}
```

## klipper version

> v0.12.0-519-gc8acb287

> Unofficial remote url: https://github.com/DangerKlippers/danger-klipper

> Repo not on offical remote/branch, expected: origin/master, detected: origin/bleeding-edge-v2

kageurufu, 2024-10-31:
The new menus haven't been merged into bleeding-edge yet.

Matszwe02, 2024-11-01:
> The new menus haven't been merged into bleeding-edge yet.

Thanks. I thought bleeding-edge-v2 is the most recent, thus not being behind in commits, thanks for the clarification. Are there plans to merge master's commits into bleeding-edge-v2, or merge some features from bleeding-edge to master?

rogerlz, 2024-11-22:
It is in sync now.. pls reopen or ping me on discord if you still have issues.

closed 2024-11-22 by rogerlz