KalicoCrew/kalico #430: Feature Request: pressure advance values based on acceleration for that move

MongooseTNM, 2024-11-12:
Title is pretty self explanatory but it would be nice if you could tune a few different acceleration PA values and have the PA automatically adjust based on the acceleration of the move. 
I see 2 ways of doing this. 

You could have the set PA command be a matrix with acceleration values and PA values. Then klipper could interpolate the values in between the define ones

or we could figure out if the PA value is linear with acceleration and transfers over between filament types and just define a tunable constant to multiply by the acceleration and add to the base PA value. kind of like a PA per mm/s^2

mf7120, 2024-11-14:
Hey,
This feature can be enabled in OrcaSlicer, see latest realease and "Adaptive PA" in the filament section.

MongooseTNM, 2024-11-17:
I dont use orcaslicer

HeavilyModdedEnder3, 2024-12-30:
This would be helpful for machines using limited kinematics, as OrcaSlicer cannot support those.

rogerlz, 2025-05-15:
Also, to change the config/default acceleration you need to use `SET_VELOCITY_LIMIT`, so why don’t you also use `SET_PRESSURE_ADVANCE` at the same time?