KalicoCrew/kalico #434: ADXL345 в роли Z-endstop

AlatusAles, 2024-11-26:
Приветствую!
Я очень впечатлён вашей работой!
Сейчас строю новый принтер (хобби) и нашел [этот "выпуск"](https://github.com/Klipper3d/klipper/issues/3741) (это великолепная работа!).
Я осознаю все риски, но конфигурация и материалы комплектующих моего принтера хорошо подходят для использования ADXL345 в качестве virtual-endstop. 
Прошу, добавить эту возможность, в качестве тестируемой.
Спасибо!

rogerlz, 2024-11-26:
sorry, but I can only read English or Portuguese.

AlatusAles, 2024-11-26:
> Извините, но я умею читать только на английском и португальском языках.

Hello!
I am very impressed with your work!
I am currently building a new printer (hobby) and found this [issue](https://github.com/Klipper3d/klipper/issues/3741) (it is great work!).
I understand all the risks, but the design and materials of my printer components are well suited for using the ADXL345 as a virtual endpoint.
Please add this feature as a testable one.
Thanks!

rogerlz, 2025-01-21:
Hi, 

I don't believe we have anyone on the team currently working on this, and it doesn't seem popular enough.
I have received feedback from the community that this 3rd party module works well: https://github.com/jniebuhr/adxl345-probe

closed 2025-01-21 by rogerlz