KalicoCrew/kalico #439: Python gcode_macro doesn't work

Gethe, 2024-11-29:
Maybe I missed something, but I was trying out the new python gcode_macro feature and I kept getting errors when running them. As a sanity check, I tried the [EXTRUDER_TEMP](https://dangerklipper.io/Command_Templates.html#python-printer-objects) example macro:
```gcode
[gcode_macro EXTRUDER_TEMP]
gcode:
    !ACTUAL_TEMP = printer["extruder"]["temperature"]
    !TARGET_TEMP = printer["extruder"]["target"]
    !
    !respond_info("Extruder Target: %.1fC, Actual: %.1fC" % (TARGET_TEMP, ACTUAL_TEMP))
```

and I get this in the console output:
```
15:30:26 $ EXTRUDER_TEMP
15:30:26 // Unknown command:"ACTUAL_TEMP="
15:30:26 // Unknown command:"TARGET_TEMP="
15:30:26 // Unknown command:"RESPOND_INFO(""
```

This is on commit 5f449f0109f51c7b7b4425b7c269f1c7c7ab3897

rogerlz, 2024-12-02:
Hi, I couldn't reproduce and I have just added a test case to cover it (#440). You are probably missing a klippy restart. 
Feel free to reopen with more details.

closed 2024-12-02 by rogerlz

Gethe, 2024-12-03:
I did multiple FIRMWARE_RESTARTs while testing it out, but I'll try it again when I get home.