KalicoCrew/kalico #445: Bug with canbus Query

TessaKavanagh, 2024-12-03:
Hi, i flashed an STM32G0B1 (StealthMax2 PCB) with Katapult first, after that i flashed Klipper firmware. After flashing the "~/klippy-env/bin/python ~/klipper/scripts/canbus_query.py can0" was reporting Application CanBoot.

I could not get it report "Klipper" even after several trys. Even after flashing Klipper (without Katapult) the reporting showed CanBoot.

After a lot of help and trying katapults "python3 flashtool.py -i can0 -q" i realised that this one is reporting "katapult" where "~/klippy-env/bin/python ~/klipper/scripts/canbus_query.py can0" is showing CanBoot

Klipper v0.12.0-520-g7aa9f3c0
[image]

Drachenkaetzchen, 2024-12-03:
I was helping Tessa with figuring this out, it seems that there is a bug converting the application ID to the proper name. 

The board is in fact running in Klipper Mode, this is purely a display error.

kageurufu, 2024-12-03:
The raw message received is `20bc9c1a451c5c11`

`0x20` is "RESP_NEEDS_NODEID" (e.g., Unassigned)
`0x11` is CanBoot/Katapults APP_ID

I'm sending `0x11` instead of `0x07` 🙃

PR #449 (2024-12-03): Fix the canserial query response
I was accidentally sending the CanBoot/Katapult app_id 0x11 instead of the Danger app_id 0x07

Additionally, its Katapult now not CanBoot

Fixes #445 

## Checklist

- [ ] pr title makes sense
- [ ] squashed to 1 commit
- [ ] added a test case if possible
- [ ] if new feature, added to the readme
- [ ] ci is happy and green
closed 2024-12-03 by rogerlz