KalicoCrew/kalico #467: Unknown config object 'danger_options'

Derkwin, 2024-12-06:
Hello, I have this error and I don’t know how to correct it. I don’t have access to Discord because the invitation is declined so I’m trying here because I can’t find anyting with google.

BoredFishRE, 2024-12-06:
could you send your config file? would help a lot

Derkwin, 2024-12-07:
Sorry for the delay here my log. hope I attach it correctely.

[klippy.log](https://github.com/user-attachments/files/18048281/klippy.log)

Derkwin, 2024-12-08:
Dont know  what happen by after I closed my printer and restart the error have been replace by this one: Option 'home_current' is not valid in section 'tmc5160 stepper_x' , here my config.
[config-20241208-081929.zip](https://github.com/user-attachments/files/18052259/config-20241208-081929.zip)

closed 2024-12-08 by Derkwin