KalicoCrew/kalico #513: FR: Nozzle probing for eddy sensor probes

HeavilyModdedEnder3, 2024-12-30:
This would greatly help z-offset issues with probes, like BTT Eddy, as shown by Beacon Contact.

recrudesce, 2024-12-30:
The Contact probing capability of the Beacon isn't handled by Klipper/Kalico, but by the firmware on the probe itself.  If the Eddy doesn't have that capability in its firmware, then it's not possible to do Contact based probing with it.

HeavilyModdedEnder3, 2025-01-07:
[https://github.com/KevinOConnor/klipper-dev/tree/work-ldctap-20240430](url)
Seems like this can be in the main branch.

Zeanon, 2025-01-07:
Just cause there is a pic doesn't mean it's working well or reliably without destroying build plates.
Ask Matt how much work went into making beacon contact stable.

HeavilyModdedEnder3, 2025-01-07:
[https://klipper.discourse.group/t/tap-detection-with-ldc1612-inductive-sensors/16234/10](url) Another source. Beacon had to make contact with no previously done contact eddy probes. This isn't new ground anymore, so it should be easier.

Zeanon, 2025-01-07:
You are of course welcome to pr the code for it.
That being said:
1. Beacon contact code is quite different from what you proposed as it is done in the beacon firmware
2. Beacon firmware is not open source so it is still new ground
3. I still think nozzle probing in klippy is a bad idea due to latency and python being python 
4. The current Eddy implementation is just plain bad

HeavilyModdedEnder3, 2025-01-07:
Cartographer probing should be open source

yell3D, 2025-01-07:
cartographer's scanner.py is forked from beacon.py and therefore use the same starting point. Both are incompatible with BTT's Eddy probe. 

If YOU are up the game to create code to add to Kalico you may look at https://github.com/Cartographer3D/OpenBedScanner which is for rp2040 bed scanner.

Honestly just buy beacon or carto and call it a day.

recrudesce, 2025-01-08:
> Cartographer probing should be open source

The firmware isn't though, which is what does the nozzle probing aspect.  The Klipper module does NOT handle the nozzle probing in either case (Beacon or Cartographer).

HeavilyModdedEnder3, 2025-01-30:
#546

closed 2025-01-30 by HeavilyModdedEnder3