KalicoCrew/kalico #528: Error printing with secondary extruder due to Pressure Advance

GrafEffvonX, 2025-01-09:
I'm running a custom toolchanger printer with two toolheads that share one extruder stepper motor.
For this I'm using [TypQxQ/KTC](https://github.com/TypQxQ/KTC) for the toolchanging and [SYNC_EXTRUDER_MOTION ](https://www.klipper3d.org/G-Codes.html#sync_extruder_motion) to sync the hardware stepper motor to the secondary extruder that doesn't have a stepper defined.

This leads to the following error when executing a print-move with the second toolhead:
```
Internal error on command:"G1"
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/gcode.py", line 309, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/extras/gcode_move.py", line 183, in cmd_G1
    self.move_with_transform(self.last_position, self.speed)
  File "/home/pi/klipper/klippy/extras/bed_mesh.py", line 274, in move
    self.toolhead.move(split_move, speed)
  File "/home/pi/klipper/klippy/toolhead.py", line 597, in move
    self.lookahead.add_move(move)
  File "/home/pi/klipper/klippy/toolhead.py", line 233, in add_move
    self.flush(lazy=True)
  File "/home/pi/klipper/klippy/toolhead.py", line 221, in flush
    self.toolhead._process_moves(queue[:flush_count])
  File "/home/pi/klipper/klippy/toolhead.py", line 455, in _process_moves
    self.extruder.move(next_move_time, move)
  File "/home/pi/klipper/klippy/kinematics/extruder.py", line 324, in move
    pressure_advance = self.extruder_stepper.pressure_advance
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'pressure_advance'
Transition to shutdown state: Internal error on command:"G1"
 ```

this is the config for the extruder1 without a motor:
 ```
[extruder1] #sync with extruder0
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: KTC_1_HEATER
sensor_type: Generic 3950 # dragon
sensor_pin: KTC_1_SENSOR
control: pid
pid_Kp: 21.776
pid_Ki: 0.937
pid_Kd: 126.570
min_temp: 0
max_temp: 300
min_extrude_temp: 10
 ```
I don't know the correct way to fix this, as Klipper doesn't even allow me to specify `pressure_advance: ` on this second extruder (i don't know if this is because there is no motor, but i suspect that to be the case).

It would be nice if kalico would enable pressure_advance on extruders without stepper motors and I'll look into getting that working and maybe even make a pull request, if i can get that working. The other option would be to just use the pressure advance from the first extruder or just don't use pressure advance for the second toolhead at all.

rogerlz, 2025-01-21:
Thanks for reporting it. I believe I have a fix in https://github.com/KalicoCrew/kalico/pull/548, but I can't test it right now.

ksummers92, 2025-01-31:
I would also like to add that I believe that this is causing a similar issue when using Happy Hare for MMU control. I get a "Active Extruder does not have a stepper" error. Does not appear to be an issue on Kalico/main, but it does appear in bev2. So might be a similar but different issue.

closed 2025-02-03 by rogerlz