KalicoCrew/kalico #532: Enhancement : Support more than 3 axis for kinematics

JordyDH, 2025-01-10:
Hi,

One of the main limitations preventing Klipper from being used for broader applications is its lack of support for multiple axes beyond the standard three.

I've been working with [naikymen's fork for CNC and pipetting applications](https://github.com/naikymen/klipper-for-cnc/tree/pipetting) and contributed to identifying and resolving issues that expanded Klipper's capabilities to support 9 axes. I can confidently say that Klipper handles this load exceptionally well, with all axes moving synchronously across multiple MCUs.

Given this success, I’m curious if there is any interest from the Kalico team in officially expanding motion support beyond 3 axes.

Looking forward to your thoughts!

naikymen, 2025-01-10:
I'd be glad to help with this proposal. I think joining efforts is important too. :)

I would need to learn a bit more about how that should happen, and surely help from the kalico crew if they're interested.

Saludos!
Nico

naikymen, 2025-01-10:
For anyone interested, here is some discussion: https://discord.com/channels/1297243471442214913/1314324665380110377/1327192862563237888

rogerlz, 2025-02-28:
Hi, 

I, personally, have been following naikymen's for a while and hope to play with it sometime soon, but I do not think it fits Kalico at the moment. 

I am closing this issue for now since we have a duplicate feature request in our discord: https://discord.com/channels/1297243471442214913/1325478337820364800

closed 2025-02-28 by rogerlz