KalicoCrew/kalico #575: Bed MPC cooling fan doesn't work with multi_pin setup

wight554, 2025-02-15:
I've got filter and bed fans connected to different ports and combine them multi_pin. When I tried to setup bed mpc using that fans, I got error.

Example configuration:

```
[multi_pin multiple_filter_fans]
pins: FILTER_FAN, BED_FAN

[fan_generic chamber_fans]
pin: multi_pin:multiple_filter_fans

[heater_bed]
control: mpc
heater_power: 75
cooling_fan: fan_generic chamber_fans
```

Commenting out this line solves the problem

```
cooling_fan: fan_generic chamber_fans
```

Error:

```
Unknown pin chip name 'multi_pin'
```

wight554, 2025-02-15:
I haven't tried extruder, might be affected as well

rogerlz, 2025-02-28:
I tested this myself to write the discord #fac (frequently asked configs) so I am sure it works. The multi_pin usage requires a specific ordering in the config file. Please test again and reopen if this is till the case with your printer.cfg/klippy.log attached.

closed 2025-02-28 by rogerlz

wight554, 2025-04-13:
@rogerlz pinged you in discord, but I'll duplicate here for tracking. Unfortunately I can't reopen issue

Discord FAC has the following config example:
```ini
[multi_pin bedfan_pins]
pins: PA4, PA5

[heater_fan bed_fan]
pin: multi_pin:bedfan_pins
heater: heater_bed
heater_temp: 50

[heater_bed]
heater_pin: PA2
sensor_type: Generic 3950
sensor_pin: PF3
min_temp: 10
max_temp: 130
control: mpc
cooling_fan: heater_fan bed_fan
heater_power: 650
```

That example uses heater_fan which is not something you can turn off/on/pwm so cooling_fan section probably has zero effect. I've tried to switch fan_generic to heater_fan and issue disappeared but it's not controllable anymore.

I'll also attach my printer config so you can see order is correct, it also works when used for macro below

[printer.cfg.txt](https://github.com/user-attachments/files/19727228/printer.cfg.txt)