KalicoCrew/kalico #582: tmc2209's sense_resistor is required, but is undocumented with no default value

synthead, 2025-02-25:
Related to https://github.com/KalicoCrew/kalico/issues/325.

I copied my configs from Klipper, and when starting Kalico, I am seeing this error (Kalico is started as a systemd service on my host):

```
Feb 24 23:06:51 enderng python[1169]: ERROR:root:Config error
Feb 24 23:06:51 enderng python[1169]: Traceback (most recent call last):
Feb 24 23:06:51 enderng python[1169]:   File "/usr/lib/kalico/klippy/printer.py", line 287, in _connect
Feb 24 23:06:51 enderng python[1169]:     self._read_config()
Feb 24 23:06:51 enderng python[1169]:     ~~~~~~~~~~~~~~~~~^^
Feb 24 23:06:51 enderng python[1169]:   File "/usr/lib/kalico/klippy/printer.py", line 215, in _read_config
Feb 24 23:06:51 enderng python[1169]:     self.load_object(config, section_config.get_name(), None)
Feb 24 23:06:51 enderng python[1169]:     ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Feb 24 23:06:51 enderng python[1169]:   File "/usr/lib/kalico/klippy/printer.py", line 199, in load_object
Feb 24 23:06:51 enderng python[1169]:     self.objects[section] = init_func(config.getsection(section))
Feb 24 23:06:51 enderng python[1169]:                             ~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Feb 24 23:06:51 enderng python[1169]:   File "/usr/lib/kalico/klippy/extras/tmc2209.py", line 97, in load_config_prefix
Feb 24 23:06:51 enderng python[1169]:     return TMC2209(config)
Feb 24 23:06:51 enderng python[1169]:   File "/usr/lib/kalico/klippy/extras/tmc2209.py", line 63, in __init__
Feb 24 23:06:51 enderng python[1169]:     current_helper = tmc2130.TMC2130CurrentHelper(config, self.mcu_tmc)
Feb 24 23:06:51 enderng python[1169]:   File "/usr/lib/kalico/klippy/extras/tmc2130.py", line 202, in __init__
Feb 24 23:06:51 enderng python[1169]:     super().__init__(config, mcu_tmc, MAX_CURRENT)
Feb 24 23:06:51 enderng python[1169]:     ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Feb 24 23:06:51 enderng python[1169]:   File "/usr/lib/kalico/klippy/extras/tmc.py", line 791, in __init__
Feb 24 23:06:51 enderng python[1169]:     self.sense_resistor = config.getfloat("sense_resistor", above=0.0)
Feb 24 23:06:51 enderng python[1169]:                           ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Feb 24 23:06:51 enderng python[1169]:   File "/usr/lib/kalico/klippy/configfile.py", line 155, in getfloat
Feb 24 23:06:51 enderng python[1169]:     return self._get_wrapper(
Feb 24 23:06:51 enderng python[1169]:            ~~~~~~~~~~~~~~~~~^
Feb 24 23:06:51 enderng python[1169]:         self.fileconfig.getfloat,
Feb 24 23:06:51 enderng python[1169]:         ^^^^^^^^^^^^^^^^^^^^^^^^^
Feb 24 23:06:51 enderng python[1169]:     ...<6 lines>...
Feb 24 23:06:51 enderng python[1169]:         note_valid=note_valid,
Feb 24 23:06:51 enderng python[1169]:         ^^^^^^^^^^^^^^^^^^^^^^
Feb 24 23:06:51 enderng python[1169]:     )
Feb 24 23:06:51 enderng python[1169]:     ^
Feb 24 23:06:51 enderng python[1169]:   File "/usr/lib/kalico/klippy/configfile.py", line 84, in _get_wrapper
Feb 24 23:06:51 enderng python[1169]:     raise error(
Feb 24 23:06:51 enderng python[1169]:     ...<2 lines>...
Feb 24 23:06:51 enderng python[1169]:     )
Feb 24 23:06:51 enderng python[1169]: configparser.Error: Option 'sense_resistor' in section 'tmc2209 stepper_x' must be specified
```

This is for a BIGTREETECH SKR Mini E3 V3.0, and I believe the resistors are permanent SMD parts.  I am unaware of what resistors are on the tmc2209 drivers, and I don't believe that it's adjustable.  Here is a photo:

[image]

Here is the tmc2209 stepper_x config section in the example config.  There is no mention of `sense_resistor` here:

https://github.com/KalicoCrew/kalico/blob/aa4cc80507a985b33de8b037cb8e400e7e168201/config/generic-bigtreetech-skr-mini-e3-v3.0.cfg#L12-L21

Additionally, this is the TMC driver documentation.  There is no mention of `sense_resistor` here, either:

https://github.com/KalicoCrew/kalico/blob/aa4cc80507a985b33de8b037cb8e400e7e168201/docs/TMC_Drivers.md?plain=1#L1-L539

Here is part of the tmc2209 documentation in docs/Config_Reference.md, as seen at https://docs.kalico.gg/Config_Reference.html#tmc2209.  The `sense_resistor` option is here, but there is no comment, description, or default value.

https://github.com/KalicoCrew/kalico/blob/aa4cc80507a985b33de8b037cb8e400e7e168201/docs/Config_Reference.md?plain=1#L4299-L4308

I'm not sure what to put here, or why.  The BIGTREETECH SKR Mini E3 V3.0 also has a soldered resistor on it, so I believe that the config example should contain the value appropriate for the board.

I did find this information on https://github.com/KalicoCrew/kalico/issues/325:

> Klipper assumes 0.075 Ω (as described in an ever so small note in the documentation that the user did not read), and configures the TMC driver accordingly.

I wasn't sure if 0.075 Ω meant that 0.075 is the default value here, but I did see that this value is supposed to be a float here:

https://github.com/KalicoCrew/kalico/blob/aa4cc80507a985b33de8b037cb8e400e7e168201/klippy/extras/tmc.py?plain=1#L783-L791

I added `sense_resistor: 0.075` to my `[tmc2209 stepper_x]` section, and this seems to have worked, but this is a guess to make the error go away.

```ini
[tmc2209 stepper_x]            
sense_resistor: 0.075          
uart_pin: PC11
tx_pin: PC10
uart_address: 0
run_current: 1.8
```

Let's update the documentation to include a comment to describe what this setting is for, and how to acquire a value.  We should also say that this is supposed to be a float that represents Ohms.

Even though https://github.com/KalicoCrew/kalico/issues/325 avoids suggesting default values, I believe that we should include the correct value for [generic-bigtreetech-skr-mini-e3-v3.0.cfg](https://github.com/KalicoCrew/kalico/blob/aa4cc80507a985b33de8b037cb8e400e7e168201/config/generic-bigtreetech-skr-mini-e3-v3.0.cfg).  This is arguably safer, because folks may accidentally use a potentially destructive value out of confusion (like me!).

CarVac, 2025-02-25:
Yeah, the documentation should tell the user to seek out the schematic if the resistor is not visible. In your case, https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V3.0/Hardware/BTT%20E3%20SKR%20MINI%20V3.0_SCH.pdf

[image]

Ataraxia-Mechanica, 2025-02-26:
TMC2209's sense resistor is usually 0.11 by the way, 0.075 is the usual value for TMC5160. This is mentioned in klipper's document:
https://www.klipper3d.org/Config_Reference.html#tmc2209

PR #590 (2025-02-26): docs: improve the tmc sense_resistor description
fixes #582