KalicoCrew/kalico #592: FR: Make Neopixel Timing values configureable

HelgeKeck, 2025-02-27:
Please make it possible to configure these two values in the printer.cfg. Ideally you can change these values with a gcode command as well to be able to tune the correct values without rebooting the whole time

[image]

rogerlz, 2025-03-10:
```diff
diff --git a/klippy/extras/neopixel.py b/klippy/extras/neopixel.py
index 4eb69be6..122e4623 100644
--- a/klippy/extras/neopixel.py
+++ b/klippy/extras/neopixel.py
@@ -9,6 +9,8 @@ BACKGROUND_PRIORITY_CLOCK = 0x7FFFFFFF00000000

 BIT_MAX_TIME = 0.000004
 RESET_MIN_TIME = 0.000050
+PULSE_SHORT_TIME = 0.000200  # Default value from neopixel.c
+PULSE_LONG_TIME = 0.000650   # Default value from neopixel.c

 MAX_MCU_SIZE = 500  # Sanity check on LED chain length

@@ -25,6 +27,11 @@ class PrinterNeoPixel:
         self.pin = pin_params["pin"]
         self.mcu.register_config_callback(self.build_config)
         self.neopixel_update_cmd = self.neopixel_send_cmd = None
+        # Get timing parameters
+        self.bit_max_time = config.getfloat('bit_max_time', BIT_MAX_TIME, above=0.)
+        self.reset_min_time = config.getfloat('reset_min_time', RESET_MIN_TIME, above=0.)
+        self.pulse_short_time = config.getfloat('pulse_short_time', PULSE_SHORT_TIME, above=0.)
+        self.pulse_long_time = config.getfloat('pulse_long_time', PULSE_LONG_TIME, above=0.)
         # Build color map
         chain_count = config.getint("chain_count", 1, minval=1)
         color_order = config.getlist("color_order", ["GRB"])
@@ -55,12 +62,15 @@ class PrinterNeoPixel:
         )

     def build_config(self):
-        bmt = self.mcu.seconds_to_clock(BIT_MAX_TIME)
-        rmt = self.mcu.seconds_to_clock(RESET_MIN_TIME)
+        bmt = self.mcu.seconds_to_clock(self.bit_max_time)
+        rmt = self.mcu.seconds_to_clock(self.reset_min_time)
+        pst = self.mcu.seconds_to_clock(self.pulse_short_time)
+        plt = self.mcu.seconds_to_clock(self.pulse_long_time)
         self.mcu.add_config_cmd(
             "config_neopixel oid=%d pin=%s data_size=%d"
             " bit_max_ticks=%d reset_min_ticks=%d"
-            % (self.oid, self.pin, len(self.color_data), bmt, rmt)
+            " pulse_short_ticks=%d pulse_long_ticks=%d"
+            % (self.oid, self.pin, len(self.color_data), bmt, rmt, pst, plt)
         )
         cmd_queue = self.mcu.alloc_command_queue()
         self.neopixel_update_cmd = self.mcu.lookup_command(
```

```diff
diff --git a/src/neopixel.c b/src/neopixel.c
index bbea09f8..f488f790 100644
--- a/src/neopixel.c
+++ b/src/neopixel.c
@@ -74,8 +74,6 @@ neopixel_delay(neopixel_time_t start, neopixel_time_t ticks)

 #endif

-#define PULSE_LONG_TICKS  nsecs_to_ticks(650)
-#define PULSE_SHORT_TICKS nsecs_to_ticks(200)
 #define BIT_MIN_TICKS     nsecs_to_ticks(1250)

@@ -86,6 +84,8 @@ neopixel_delay(neopixel_time_t start, neopixel_time_t ticks)
 struct neopixel_s {
     struct gpio_out pin;
     neopixel_time_t bit_max_ticks;
+    neopixel_time_t pulse_short_ticks;
+    neopixel_time_t pulse_long_ticks;
     uint32_t last_req_time, reset_min_ticks;
     uint16_t data_size;
     uint8_t data[0];
@@ -103,10 +103,13 @@ command_config_neopixel(uint32_t *args)
     n->pin = pin;
     n->data_size = data_size;
     n->bit_max_ticks = args[3];
+    n->pulse_short_ticks = args[5];
+    n->pulse_long_ticks = args[6];
     n->reset_min_ticks = args[4];
 }
 DECL_COMMAND(command_config_neopixel, "config_neopixel oid=%c pin=%u"
-             " data_size=%hu bit_max_ticks=%u reset_min_ticks=%u");
+             " data_size=%hu bit_max_ticks=%u reset_min_ticks=%u"
+             " pulse_short_ticks=%u pulse_long_ticks=%u");

 static int
 send_data(struct neopixel_s *n)
@@ -142,19 +145,19 @@ send_data(struct neopixel_s *n)
                 last_start = start;
                 byte <<= 1;

-                neopixel_delay(start, PULSE_LONG_TICKS);
+                neopixel_delay(start, n->pulse_long_ticks);
                 irq_disable();
                 gpio_out_toggle_noirq(pin);
                 irq_enable();

-                neopixel_delay(neopixel_get_time(), PULSE_SHORT_TICKS);
+                neopixel_delay(neopixel_get_time(), n->pulse_short_ticks);
             } else {
                 // Short pulse
                 neopixel_delay(last_start, BIT_MIN_TICKS);
                 irq_disable();
                 neopixel_time_t start = neopixel_get_time();
                 gpio_out_toggle_noirq(pin);
-                neopixel_delay(start, PULSE_SHORT_TICKS);
+                neopixel_delay(start, n->pulse_short_ticks);
                 gpio_out_toggle_noirq(pin);
                 irq_enable();
```

rcloran, 2025-11-12:
Related: https://github.com/Klipper3d/klipper/pull/7113 , https://github.com/KalicoCrew/kalico/pull/771