KalicoCrew/kalico #609: Feature Request: axis_twist_compensation multiple sensors

darkjacky, 2025-03-01:
It would be nice to be able to unplug (or splice) for example the X or Y axis stop switch then plug in a button mounted right on the nozzle to get automated and accurate readings for axis_twist_compensation. (Or instead of a button a piece of aluminium tape with a wire clamp on it and then another clamp on the bed.)
So my request is for the axis_twist_compensation to support an optional endstop_pin variable.
How it would work: We first home the printer then unplug (or splice) one of our endstops and plug in our nozzle mounted button. During this time we need access to the height of our Z axis. Then after we are done with the button we can lower the nozzle and/or press accept so the axis_twist_compensation can do its magic. It would read the BL_Touch sensor then read the nozzle sensor then continue to the next point.
This would significantly reduce the workload required for axis_twist_compensation and significantly increase the accuracy.
Also this would likely require an additional samples variable (or use the one from the bl_touch) to increase accuracy even more.
Even if this feature isn't used often (usually only once or never) I think this would be a great addition to this feature.

Thank you

ShaneDelmore, 2025-03-13:
I just decided to run axis twist compensation on my machine, thinking it would use my beacon, so I was surprised when it prompted me to do manual adjustments so I'm just here to say, yes, I agree it would be great to use the sensors we already use when they are available.

synthead, 2025-04-03:
I would love to see this.  Ultimately, I think that this feature requires two bits of work:

- Support multiple z probes
- Support two probes during axis_twist_compensation calibration

The nozzle is ultimately the source of truth.  It has a 0, 0, 0 offset from itself.  If the nozzle was able to be used as a probe, then axis twist wouldn't matter for z height since the "probe" would be perfect, always.

I experimented with adding a wire from my nozzle, and flipping over a single-sided PEI sheet (with exposed metal on the other side), like so:

[image]

[image]

Then, I used my unused z-stop sensor on my BIGTREETECH SKR MINI E3 V3.0 to connect the nozzle and the bed sheet, and used this Kalico config:

```ini
[probe]
pin: !PC2
speed: 1
lift_speed: 500
z_offset: 0
samples: 3
samples_tolerance: 0.1
samples_tolerance_retries: 5
sample_retract_dist: 0.4

[stepper_z]
endstop_pin: probe:z_virtual_endstop
homing_speed: 2
second_homing_speed: 1
homing_retract_dist: 0.5
homing_retract_speed: 10

[bed_mesh]
probe_count: 5, 5
horizontal_move_z: 1
mesh_min: 5, 5
mesh_max: 225, 225

[safe_z_home]
home_xy_position: 115, 115
```

With the above, after a heat soak, I was able to probe my bed and get this mesh:

[image]

This works pretty good, but I caution anyone reading this that nozzles get dirty, and don't always make good electrical contact.  If you have old plastic stuck to your nozzle, you may not even achieve continuity at all, and you might sink your nozzle right into your bed with force.

It would be a much better idea to use something other than the nozzle for continuity, like a bit of solder wick, a spring, or something else that's flexible, so in case contact doesn't occur when it physically touches, contact will be made after a little force has been applied that doesn't hurt the machine.

With something like the above, if we implemented an automatic twist calibration between two probes, that would be so awesome.  My current workaround is to use a continuity tester and listen for beeps as opposed to the paper test, but this isn't as good as having a probe with multiple samples and tolerance like the above config.  It is still much better than the paper test, though.