KalicoCrew/kalico #616: Dockable probe stopped working on bed slinger after aa4cc80507a985b33de8b037cb8e400e7e168201

wight554, 2025-03-13:
I use klack probe from KevinAkaSam on my ender 3 both for homina and probing with dockable probe configured using dockable probe config
After aa4cc80507a985b33de8b037cb8e400e7e168201 my klipper won't start as I have Z for approach and detach positions. I might trick that with new macros froim that commit but that'd be no joy.
I don't feel like there's a good HW way to workaround the problem on my setup, so I believe it would be better to add option to disable that error and keep it 'False' by default to prevent regression.

My config:
```
[dockable_probe]
pin: ^toolhead:PROBE_INPUT
x_offset: 0 # negative = left of the nozzle
y_offset: 40.105
check_open_attach: True
z_offset: 2.850
speed: 8
lift_speed: 15
samples: 4
samples_result: median
sample_retract_dist: 2.0
samples_tolerance: 0.01
samples_tolerance_retries: 3

# dockable
attach_speed: 150
detach_speed: 150
travel_speed: 300
approach_position: 212, 0, 20
dock_position: 234, 0
detach_position: 212, 0, 1
z_hop: 10.0
drop_first_result: True
```

Alby312, 2025-03-30:
I have the same problem, the error is:
DockableProbe cannot be used as Z endstop if a z position
is defined in approach/dock/extract/insert/detach position."

I use the detachable probe as z endstop and i require a z position to detach the probe

BildermanKawasaki, 2025-04-23:
Same problem for me and several of my friends.

rogerlz, 2025-04-24:
This has been fixed in https://github.com/KalicoCrew/kalico/pull/652
Please try again and reopen if needed.

closed 2025-04-24 by rogerlz

wight554, 2025-04-24:
All good, no more regression
Thanks