KalicoCrew/kalico #691: Kalico migration (Moonraker can't connect to klipper)

Ineedmoney420, 2025-05-14:
I have tried a few times and even complete new formatted SD cards, Used Kiauh to install Klipper, moon and Mainsail. From there I tried to migrate it to kalico and I either get an "MCU not connected" error even thought the serial ID is good in printer.cfg or Moonraker can't connect to Klipper!
Please check if the Klipper service is running and klippy_uds_address is correctly configured in the moonraker.conf. regular klipper seems to run just fine. Maybe newer klipper is changing directories ??? Not sure

Really need this for sensor less setup

Ineedmoney420, 2025-05-14:
Not the only one it appears...
https://discord.com/channels/460117602945990666/1363241838634467479/1363241838634467479

stokedgsxr, 2025-05-14:
I ran into this tonight as well.

Ineedmoney420, 2025-05-15:
Fixed moonraker connection to klipper with  **~/klippy-env/bin/pip install -r ~/klipper/scripts/klippy-requirements.txt**

Currently trying to figure out out why the port is closed to the MCU, Klippy log attached

[klippy (2).log](https://github.com/user-attachments/files/20218550/klippy.2.log)

rogerlz, 2025-05-15:
original issue has been fixed and the next is being worked on in a VD discord thread: https://discord.com/channels/460117602945990666/1363241838634467479

closed 2025-05-15 by rogerlz