KalicoCrew/kalico #696: Proposal: Extruder sync-to-input-shaper or equivalent in mainline

richfelker, 2025-05-20:
I'd like to get extruder sync to IS in mainline Kalico. In my experience it's probably the single most compelling reason to use BEv2 right now instead of mainline, but as implemented, it's interwoven with the heavy refactorization of PA and IS code that was largely for implementing the "input smoothers", and that makes the code harder to work with and unlikely to be merged into mainline.

Before I start actually working on anything on this, I see two possible ways it could be done:

1. Like in Dmitry's branch, having klippy split up E motion by which kinematic access it corresponds to, linking up the impulse pattern from the kinematic axes to somewhere `kin_extruder.c` in chelper can see it, and integrating the move queue multiple times, once for each IS impulse for each axis. This would give the same behavior, but is mildly invasive and computationally fairly costly (running it 3 times).
2. Instead, assigning a single configurable input shaper to the E axis. This would give exactly the same behavior if the X and Y shapers are the same, but diverging somewhat if they're different. (Note that Dmitry's version is already inaccurate if they're different; except on moves that are purely along one axis; otherwise it's a weighted interpolation of the two.) This would only require integration of one axis worth of impulses (so cost relative to no-sync would be scaling up by the number of impulses the shaper has, not also doubling that plus 1). It would also apply the shaper to retract/unretract moves, which may reduce extruder-induced vibrations on the toolhead and make it possible to use higher retract/unretract accelerations. And it would avoid the need for invasive changes to the python side of klippy to make it split up E motion by kinematic axis.

I'm not sure which approach I like better, so I'd like some feedback before I dive into actually doing anything.

FWIW: This proposal is kinda a prerequisite for #695 if that one is to be done in mainline rather than only BEv2. I mention it partly because that proposal makes each integration somewhat costlier if using nonlinear PA, so this might affect tradeoffs between the above two possible approaches.

CarVac, 2025-05-20:
How wrong exactly is the weighted interpolation of the pulse trains? Basically the effect of the shaper is to round off the corners on the nozzle path, and the goal of extruder sync should be to correctly extrude a constant width along the path the nozzle takes.

Another consideration is that the result of the shaper on the _real_ toolpath is not a faceted corner but something smooth, while the extruder motor is capable of faster relative acceleration without ringing but there's some smoothing from the actual extrusion process.

What would the utility of a third shaper just for E be? Just reducing computational requirements? How would you set it?

What's actually important, I feel, is to calculate how much less to extrude in any given corner given the expected junction deviation reducing the path length.

richfelker, 2025-05-20:
> How wrong exactly is the weighted interpolation of the pulse trains?

I haven't worked out the math, but it was documented that it's inexact.

> Basically the effect of the shaper is to round off the corners on the nozzle path

That's not the intended effect, though it will happen to some extent, more if you exceed appropriate accel/scv for the shaper parameters. What the shaper is trying to do is to split the velocity change up into multiple impulses separated in time so that successive impulses come roughly in sync with the expected vibration and counteract the "rebound" from the previous impulse.

> and the goal of extruder sync should be to correctly extrude a constant width along the path the nozzle takes.

Right. The goal of sync is to have it so that, once the change in velocity has been split up, the changes on the E axis are also split up to happen at the same time. Without this, you can get oscillating extrusion width that *looks just like ringing* where input shaper is suppressing ringing, because kinematic velocity is oscillating in time but the extrusion rate isn't. In a worst case, I've even seen sudden changes in velocity, like when my purge line goes from low to high kinematic speed trying to maintain constant E rate, produce a break in the extrusion line from the desync.

> What would the utility of a third shaper just for E be? Just reducing computational requirements? How would you set it?

Conceptually it's not automated sync but giving the user the ability to achieve full or approximate sync by choosing a shaper for E. I think this is a simpler model, less computationally costly and less complexity-costly in terms of interactions between different components of Kalico. If you have similar IS for X and Y, just frequencies differing by 10-15 Hz or so, choosing the midpoint for the E shaper would almost surely match well enough. But it does seem like this model would be more confusing and difficult for users than a purely automatic sync.

> What's actually important, I feel, is to calculate how much less to extrude in any given corner given the expected junction deviation reducing the path length.

This is a completely separate topic. JD is not supposed to reduce path length, especially when operated SCV values suitable for high quality. But if it does, there's really nothing we can do at the extruder-sync layer. This layer has to keep total motion of the axis constant and only make local variations that cancel out.

richfelker, 2025-05-20:
> What do you mean by "kinematic velocity"? The internal requested velocity? What is making it oscillate?

A lot of folks don't understand what input shaping is/does. I find it very informative to watch some of the demos at human-perceptible time scales using cranes with very low resonant frequencies, like this:

https://www.youtube.com/watch?v=gzBhTrHv0-c

Here, the carriage velocity is widely changing during the move. If an extruder were attached to it and extruding at constant rate, you would not be getting uniform extrusion.

One might object and say that, in this demo, it's the payload that's being stabilized, not the carriage, and that the payload velocity doesn't oscillate as much. That's true, but it still does somewhat. And the nozzle tip is somewhere in betweer the crane carriage and payload extremes; ideally it's coupled very rigidly to the toolhead, but the whole thing is subject to vibrations from the various parts attached to it, which IS is suppressing. So it's probably best to think of this as somewhere in between "extruder is the crane payload" and "extruder on the crane carriage with a mass off the side that can induce vibration back into the carriage".

> Commanded shaper motion: "faceted" speed changes (see this cdf graph for 100hz 0.15 damping ratio:)

If you're asking why I said oscillating when it looks like a monotonic faceted speed increase, consider what happens when you have both acceleration and deceleration ramps within the same shaper window.

But even without that, or if my wording was imprecise, the amount of extrusion per length of kinematic motion is going to vary when extruder is not sync'd to IS.

> (for that matter, I want a "dual zv" with two user-selectable frequencies, one of my printers has both 110hz and 160hz peaks and I can't correct both properly)

3hump_ei at 200 Hz will suppress both of those pretty much entirely. It gets a bad rep from the calibration tool making bad recommendations with the center frequency way too low and giving you laughably low max accel, but it's an excellent tool if you choose the center to be just below twice your lowest resonance.

CarVac, 2025-05-20:
> If you're asking why I said oscillating when it looks like a monotonic faceted speed increase, consider what happens when you have both acceleration and deceleration ramps within the same shaper window.

Even when the commanded acceleration oscillates, the resulting motion doesn't oscillate.

This is 100k accel to 100 mm/s and back down 0.00224s later on a 100hz resonant frequency printer, with ideal ZV.

[image]

richfelker, 2025-05-20:
Regardless of whether "oscillate" is correct, the velocity is changing at a rate that mismatches the ideal rate the planned extruder moves were intended for. And that's the real point: we're correcting for whatever the mismatch is.

However, I'm pretty sure it does actually oscillate in velocity if you don't pick super-low-stress example cases like a top speed of 100 at 100k accel. The hard cases are all acceleration-bound, i.e. no cruise period (and in the real world as well, someone who can do 100k accel is not going to be requesting 100 mm/s, more like 400+ mm/s), and include cases that do not follow trapezoidal acceleration profiles at all but instantaneous changes in velocity due to junctions. The worst case instantaneous jump in velocity happens in the perpendicular direction to the primary direction of motion when turning at a very slight angle (finely segmented curve) and can be up to 4x SCV in magnitude. With these in rapid succession you should get some interesting speed curves.

CarVac, 2025-05-20:
The issue is that we want the extrusion after the PA and any subsequent smoothing to match the actual motion of the toolhead after the shaping.

So maybe the right thing to do is to apply filtering to each axis separately with parameters appropriate to simulate each one's input-shaped impulse response, as a best guess estimate of the real toolhead motion.

That gives us the amount of actual extrusion that we want.

And then the goal of the downstream PA would be solely to figure out how to counteract the nonideal extrusion impulse response.

richfelker, 2025-05-20:
I'm pretty sure the existing extruder-sync code is doing at least roughly the right thing here, because it works. It outputs a weighted average of the base extruder position at the IS impulse time offsets, with their corresponding weights, plus the same weighted average of the PA velocity integral centered at those impulse time offsets. This is exactly what input shaper does for modifying the kinematic move position.

If you're saying it should look more like the purple line and less like the gray line, I'm not sure if that's the case. This is probably worth deeper investigation though.

The actual position of motion system elements will be somewhere between the gray and the purple, depending on how rigidly they're coupled. On an imaginary infinitely rigid machine, they would all be exactly like the gray, but even on a well-constructed real world machine I'd expect the point of extrusion to be at least somewhat closer to the gray, and the flimsier parts of the motion system to be all the way purple. It might make sense to have a parameter to allow tuning between these extremes for the path the extruder should follow. I think rather than trying to simualte the response to IS, the PA base position smoothing (which I previously removed) with a suitable smoothing window (not necessarily the same one you use to make PA moves realizable) would be a very, very close approximation and probably suffice. Maybe someone could run some tests and see how close they are, and figure out the right correspondence between the IS filter width and smooth-time.

richfelker, 2025-05-20:
> If you actually get motion "somewhere between the gray and the purple" then you've chosen the wrong input shaper parameters.

Again look at the crane demos as an analogy. You can clearly see the carriage follow the gray line. Does this mean they chose the wrong input shaper/parameters?

Answer 1: No, because the path they care about is the payload not the carriage. But let's pretend they care about the carriage. So now,

Answer 2: No, because in order not to have an unbounded reactive force from the payload swinging, which will eventually get strong enough to make the carriage deflect, the payload reasonance must be suppressed. And unless the carriage also has a resonance that's being excited by the motions, you can ignore it.

I claim case 2 is actually pretty close to what we're dealing with in printers, at least conditionally depending on the construction. In an extreme case (try not to cringe at this hypothetical printer), suppose you have 12mm belts, high torque motors, 16T pulleys, light TZ type hotend, and bowden, but your toolhead also has some 4028 fans hanging off the sides. The hotend almost surely follows the gray path, but the fans are on the purple path (or the orange one without IS). You need to suppress their resonance or they'll vibrate violently and eventually deflect the whole toolhead. But you're not getting the purple toolpath.

Now, most real-world printers aren't ridiculous like that. But I also don't think they're at the other end of the spectrum.

I suppose this is testable with an accelerometer if someone knows how to setup the test...

CarVac, 2025-05-20:
I see what you're saying.

That too could be simulated numerically, have the toolhead-to-nozzle be a low mass rigid body with a high spring constant for the belts, and a relatively high mass body attached to that with a moderate spring constant.

I suspect that in a resonance test, you'll find the fans hanging off the side will produce low-amplitude side peaks. The question is what the real toolhead motion looks like when you slap 2hump or 3hump on it to eliminate all the resonances...

rogerlz, 2026-10-21:
Thanks for the detailed proposal and the discussion here.

We don't plan to bring bleeding-edge-v2 features into main. BEv2 stays the place for that work, and we'll try to keep it in sync with main more often so it's easier to run day to day.

Closing this for now.

closed 2026-10-21 by rogerlz