KalicoCrew/kalico #706: Add debounce_delay as an option to filament_switch_sensor

Arkelic, 2025-06-19:
Klipper recently [added](https://github.com/Klipper3d/klipper/commit/272e815522b0bc8e0806e052b73a5cc1af979cd7) debounce_delay as an option to [filament_switch_sensor](https://www.klipper3d.org/Config_Reference.html?h=filament#filament_switch_sensor). I would like this functionality in Kalico as I have a noisy filament switch and would like to delay calling any gcode till the state of the switch stabilizes for a set amount of time

Kramax, 2025-08-19:
I implemented changes to support this by migrating the klipper changes to kalico in pull request #726 ([https://github.com/KalicoCrew/kalico/pull/726](url)).

Arkelic, 2025-08-24:
Thank you! Looking forward to it

dalegaard, 2025-10-15:
Fixed in #726 .

closed 2025-10-15 by dalegaard