KalicoCrew/kalico #714: "PWM pin max duration too large" error

yonkiman, 2025-07-07:
Just switched from generic klipper to kalico (v0.12.0-712).  Config files that work perfectly before now generate a

```
PWM pin max duration too large

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted
```

error message.   I traced it down to my bed heater section.  If this section is in printer.cfg:
```
[heater_bed]
heater_pin: PA1 
sensor_type: Generic 3950
sensor_pin: PB1
control: pid
pid_kp: 58.437
pid_ki: 2.347
pid_kd: 363.769
min_temp: 0
max_temp: 120
```
I get the error.  If I comment it out, klipper launches with no errors.  I tried messing with some of the [heater_bed] settings, but couldn't find anything.   (I had high hopes for `pwm_cycle_time`, but setting it to 0.1 or 0.01 led to the same result as not including it at all.)

I found [one other report of this issue](https://www.answeroverflow.com/m/1385691314867277865), and one other one [in Japanese](https://m.dcinside.com/board/3dprinting/132349), both with no solutions.  The first includes links to log files.   

I just switched back to generic klipper and can confirm that the above heater settings are working normally again.

fbeauKmi, 2025-07-08:
In this case, flash your board with Kalico firmware. And let us now

yonkiman, 2025-07-08:
Oh, hadn't considered that the firmware could be different.  

I recompiled under Kalico, reflashed, reinstalled the latest (Kalico FIRMWARE_VERSION:v0.12.0-715-g91fd6480), but still get the same error.

yonkiman, 2025-07-08:
And in case it's helpful, the version of Klipper that works is v0.13.0-154-g9346ad191, which is apparently compatible with the Kalico firmware (I didn't flash back to Klipper).

fbeauKmi, 2025-07-09:
I would suggest to join Kalico discord server https://kalico.gg/discord  and open an help thread with description of your hardware and  klippy.log

TheMightyGinger4, 2025-09-16:
@fbeauKmi I am reporting this issue as well. I just troubleshooted this extensively with the knowledgeable folks in Voron discord, and we were unable to resolve the error.

I'm using a Octopus Pro v1.1 board which is a STM32723 chipset. We observed that the clock frequency was reporting CLOCK_FREQ=520000000 even though we validated the correct 25mhz crystal in make menuconfig. 

I keep receiving the same error as the original report here.
_PWM pin max duration too large

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted_

We have validated that there are no references to pulse width or cycle time parameters in my config files. We completely removed an LED module that called pwm & cycle_time just to be safe.

I have attached a klippy log that shows the issue. 

Unfortunately, I have no choice but to downgrade to vanilla klipper until we can figure this out. Hope to hear from you. Thanks!

[klippy (7).log](https://github.com/user-attachments/files/22371717/klippy.7.log)

gueee, 2026-01-01:
I opened PR #791 to address this by reducing `MAX_HEAT_TIME` from 5.0s to 3.0s (avoids PWM max_duration tick overflow on high CLOCK_FREQ MCUs like 520MHz STM32H723): https://github.com/KalicoCrew/kalico/pull/791